Xilinx/CHaiDNN #91: CHaiDNN is not using double pumping by default

salcanmor, 2018-10-26:
To use double pumping, we use a function called __builtin_mac6x2_mac8x1, which uses a double frequency clock called ap_clk_div2.

According to header of XiConvolutionTop function, we can see:
```
#ifdef __SDSVHLS__
				, bool ap_clk_div2 ) {
```
`__SDSVHLS__` symbol depends on `__HLS_SYN__` and none of them are defined by default.

So, how is possible CHaiDNN use double pumping having this configuration ?

nkpavan, 2018-10-26:
@salcanmor ,

The __HLS_SYN__ flag is defined by Vivado HLS so user need not define that.

salcanmor, 2018-10-26:
As you can see in the picture below, `__HLS_SYN__` is not defined by default.

[image]

nkpavan, 2018-10-26:
@salcanmor ,
The macro __HLS_SYN__ will be enabled during HLS synthesis and will enable the double-pumping. You can go through the generated RTL and see that the propagation of 2x clock.

nkpavan, 2018-10-26:
You can find it at <project_path>/build/_sds/vhls/XiConvolutionTop/solution/impl/

closed 2018-10-26 by nkpavan