Xilinx/CHaiDNN #92: Build software fails: cannot find -lxlnxdnn

saraballeri, 2018-10-26:
Hi,
I successfully build hardware for zcu102, but when i build the software stack I get this error:

`Compiling Examples 

make -j4 -f example.mk
make[1]: Entering directory '/home/user/CHaiDNN2/software/scripts'
-e 
Compiling ResNet50

aarch64-linux-gnu-g++ -std=c++11 -D__SDSOC=1 -Wno-write-strings -D__APIMODE=1 -DXI_DIET_CHAI_Z=0 -DXI_DIET_CHAI_ZUPLUS=0 -D__POOL_ENABLE__=1 -D__DECONV_ENABLE__=1 -L../../SD_Card/protobuf/arm64/lib -I../../SD_Card/protobuf/arm64/include -L../../SD_Card/opencv/arm64/lib -I../../SD_Card/opencv/arm64/include -I../../SD_Card/opencv/arm64/include -I/opt/Xilinx/SDx/2018.2/target/aarch64-linux/include -I/opt/Xilinx/SDx/2018.2/../../Vivado/2018.2/include -L../../SD_Card/lib -L../../SD_Card/cblas/arm64/lib -lprotobuf -lpthread -lxstack -lxlnxdnn -lparser_arm -lopenblas -lopencv_core -lopencv_imgproc -lopencv_imgcodecs -ldl -lrt -llzma -ltiff -lpng16 -lz -ljpeg -lwebp ../example/resnet50_ex.cpp -o resnet50.elf
-e 
Compiling GoogleNet 

-e 
Compiling AlexNet-FCN

aarch64-linux-gnu-g++ -std=c++11 -D__SDSOC=1 -Wno-write-strings -D__APIMODE=1 -DXI_DIET_CHAI_Z=0 -DXI_DIET_CHAI_ZUPLUS=0 -D__POOL_ENABLE__=1 -D__DECONV_ENABLE__=1 -L../../SD_Card/protobuf/arm64/lib -I../../SD_Card/protobuf/arm64/include -L../../SD_Card/opencv/arm64/lib -I../../SD_Card/opencv/arm64/include -I../../SD_Card/opencv/arm64/include -I/opt/Xilinx/SDx/2018.2/target/aarch64-linux/include -I/opt/Xilinx/SDx/2018.2/../../Vivado/2018.2/include -L../../SD_Card/lib -L../../SD_Card/cblas/arm64/lib -lprotobuf -lpthread -lxstack -lxlnxdnn -lparser_arm -lopenblas -lopencv_core -lopencv_imgproc -lopencv_imgcodecs -ldl -lrt -llzma -ltiff -lpng16 -lz -ljpeg -lwebp ../example/googlenet_ex.cpp -o googlenet.elf
-e 
Compiling SSD

aarch64-linux-gnu-g++ -std=c++11 -D__SDSOC=1 -Wno-write-strings -D__APIMODE=1 -DXI_DIET_CHAI_Z=0 -DXI_DIET_CHAI_ZUPLUS=0 -D__POOL_ENABLE__=1 -D__DECONV_ENABLE__=1 -L../../SD_Card/protobuf/arm64/lib -I../../SD_Card/protobuf/arm64/include -L../../SD_Card/opencv/arm64/lib -I../../SD_Card/opencv/arm64/include -I../../SD_Card/opencv/arm64/include -I/opt/Xilinx/SDx/2018.2/target/aarch64-linux/include -I/opt/Xilinx/SDx/2018.2/../../Vivado/2018.2/include -L../../SD_Card/lib -L../../SD_Card/cblas/arm64/lib -lprotobuf -lpthread -lxstack -lxlnxdnn -lparser_arm -lopenblas -lopencv_core -lopencv_imgproc -lopencv_imgcodecs -ldl -lrt -llzma -ltiff -lpng16 -lz -ljpeg -lwebp ../example/alexnetfcn_ex.cpp -o alexnetfcn.elf
aarch64-linux-gnu-g++ -std=c++11 -D__SDSOC=1 -Wno-write-strings -D__APIMODE=1 -DXI_DIET_CHAI_Z=0 -DXI_DIET_CHAI_ZUPLUS=0 -D__POOL_ENABLE__=1 -D__DECONV_ENABLE__=1 -L../../SD_Card/protobuf/arm64/lib -I../../SD_Card/protobuf/arm64/include -L../../SD_Card/opencv/arm64/lib -I../../SD_Card/opencv/arm64/include -I../../SD_Card/opencv/arm64/include -I/opt/Xilinx/SDx/2018.2/target/aarch64-linux/include -I/opt/Xilinx/SDx/2018.2/../../Vivado/2018.2/include -L../../SD_Card/lib -L../../SD_Card/cblas/arm64/lib -lprotobuf -lpthread -lxstack -lxlnxdnn -lparser_arm -lopenblas -lopencv_core -lopencv_imgproc -lopencv_imgcodecs -ldl -lrt -llzma -ltiff -lpng16 -lz -ljpeg -lwebp ../example/ssd_ex.cpp -o ssd.elf
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/bin/../lib/gcc/aarch64-linux-gnu/7.2.1/../../../../aarch64-linux-gnu/bin/ld: cannot find -lxlnxdnn
collect2: error: ld returned 1 exit status
example.mk:36: recipe for target 'ssd.elf' failed
make[1]: *** [ssd.elf] Error 1
make[1]: *** Waiting for unfinished jobs....
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/bin/../lib/gcc/aarch64-linux-gnu/7.2.1/../../../../aarch64-linux-gnu/bin/ld: cannot find -lxlnxdnn
collect2: error: ld returned 1 exit status
example.mk:26: recipe for target 'alexnetfcn.elf' failed
make[1]: *** [alexnetfcn.elf] Error 1
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/bin/../lib/gcc/aarch64-linux-gnu/7.2.1/../../../../aarch64-linux-gnu/bin/ld: cannot find -lxlnxdnn
collect2: error: ld returned 1 exit status
example.mk:46: recipe for target 'resnet50.elf' failed
make[1]: *** [resnet50.elf] Error 1
/opt/Xilinx/SDK/2018.2/gnu/aarch64/lin/aarch64-linux/bin/../lib/gcc/aarch64-linux-gnu/7.2.1/../../../../aarch64-linux-gnu/bin/ld: cannot find -lxlnxdnn
collect2: error: ld returned 1 exit status
example.mk:21: recipe for target 'googlenet.elf' failed
make[1]: *** [googlenet.elf] Error 1
make[1]: Leaving directory '/home/user/CHaiDNN2/software/scripts'
Makefile:46: recipe for target 'all' failed
make: *** [all] Error 2`

How can I fix it? I try to search online the library but I did not find anything.

Thanks,
Sara

NeutrinoXY, 2018-10-26:
Hi,

design/build/Makefile:

# Copy lib to SD_Card dir
copy:
	cp sd_card/${LIBRARY} ../../SD_Card/lib/
	cp sd_card/BOOT.BIN ../../SD_Card/
	cp sd_card/image.ub ../../SD_Card/

saraballeri, 2018-10-26:
Hi,
I just fix the error.
In the guide of build software stack you suggest to run `make ultraclean` before run `make clean`. The command make ultraclean removes `*.so` and then it removes also libxlnxdnn.
I close the issue.
Thank you anyway.
Sara

closed 2018-10-26 by saraballeri