Xilinx/CHaiDNN #94: setup_chai_tools.sh error to run XportDNN.pyc

adu81020799, 2018-10-26:
I wanted to generated the prototxt using XportDNN  For which I ran the setup_chai_tools.sh but it gave me the following error:./setup_chai_tools.sh: line 21: ./chaidnn_tools_ENV/bin/activate: No such file or directory

smenon009, 2018-10-29:
Hi @adu81020799 
Have you tried with a fresh download? of the tools.zip? 
Also, could you check if you have the virtualenv package installed? 

We find a lot of users successfully using this tool. Please let us know if you got around this problem already.
Cheers!

adu81020799, 2018-10-29:
Hello,
Ya i  have tried it with the fresh download and virtualenv is installed currently.
this is the result of my which virtualenv command
/home/uncc_embed/anaconda2/bin/virtualenv.
I see the packages which are in the requirement  file are the caffe based dependencies. Can I use anaconda to create a virtual environment  which I have already created for the caffe? 
Anyone who has used anaconda to create a virtual environment can please help.
Regards
adarsh

smenon009, 2018-10-30:
Ok @adu81020799 . We haven't verified this tool with anaconda virtualenv. 
Yes, It would be good to hear from other CHai users who might have used anaconda distribution.

adu81020799, 2018-10-30:
Oka. I think there was a problem with my environment set up, somehow it was not allowing  me to create the Chai DNN  tools virtual environment. So to solve this issue I separately  created virtual environment and installed all the dependecies there and then copy pasted the environment files to chai DNN tools folder and then relaunch the last command in the setup.sh.It created the virtual environment for me and I could run the quantization tools.

closed 2018-10-30 by adu81020799