Xilinx/CHaiDNN #102: output SSD text file

shinya321, 2018-11-07:
Hi,
I use zcu102 and tried to execute 8bit VGG_SSD_300 by following Section 'Using Pre-built binaries' in https://github.com/Xilinx/CHaiDNN.
After executing command as below, probably I got success,
<pre>
./zcu102/ssd.elf Xilinx 8 ./images/camel.jpg ./images/goldfish.JPEG
</pre>
And I got 2 files in out directory named 'out_0_0.txt' & 'out_1_0.txt'.

But I am sorry I cannot understand what this file means, so could you please tell me??
the example is as below:

_out_0_0.txt_
<pre>
20.000000
0.000000
17.000000
0.750853
0.094001
0.136513
0.884444
0.971263
0.000000
10.000000
0.269514
0.075035
0.150173
0.918462
0.978406
0.000000
  :
  :
17.000000
0.010036
0.945724
0.891225
0.981854
0.923620
</pre>

Best regards,
shinya321

salcanmor, 2018-11-07:
For the example: https://github.com/Xilinx/CHaiDNN/blob/master/SD_Card/images/fish.jpg

200.000000 -> Number of objects detected

0.000000 -> Beginning of object number 1

15.000000 -> label, in this case it is 'person'

0.999833 -> probability

0.425969 -> xmin

0.011450 -> ymin

0.714902 -> xmax

0.521883 -> ymax

0.000000 -> Beginning of object number 2

2.000000 -> label, 'bicycle'

0.603002 -> probability

0.134202

0.212187

0.924645

0.964356

0.000000 -> Beginning of object number 3

15.000000

0.163214 -> probability is very low it will not be painted in the image.

0.370299

0.075211

0.781649

0.877746

shinya321, 2018-11-08:
Dear salcanmor,
Thank u very much for your great information!!
I could completely understand what it means.

closed 2018-11-08 by shinya321