Xilinx/CHaiDNN #104: Fail to build CHaiDNN on ZCU102 platform

sheneuler, 2018-11-11:
Hello,
I got issue when I was trying to build CHaiDNN version 2 on ZCU102 platform on SDSoC 2018.2.

===>The following messages were generated while  Compiling (top level synthesis checkpoint) dynamic region Log file: D:/SourceCode/2018/AI/CHaiDNN/design/build/_sds/p0/vivado/prj/prj.runs/synth_1/runme.log  :
ERROR: [VPL-1] couldn't read file "D:SourceCode?8AICHaiDNNdesignuild/../conv/scripts/mcps.tcl": no such file or directory
ERROR: [VPL 60-704] Integration error, One or more synthesis runs failed during dynamic region dcp generation
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling 'D:/Xilinx/SDx/2018.2/bin/vpl   --iprepo D:/SourceCode/2018/AI/CHaiDNN/design/build/_sds/iprepo/repo  --iprepo D:/Xilinx/SDx/2018.2/data/ip/xilinx  --platform C:/Users/ASUS/AppData/Roaming/Xilinx/Vivado/zcu102/zcu102.xpfm  --temp_dir D:/SourceCode/2018/AI/CHaiDNN/design/build/_sds/p0  --output_dir D:/SourceCode/2018/AI/CHaiDNN/design/build/_sds/p0/vpl  --input_file D:/SourceCode/2018/AI/CHaiDNN/design/build/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels XiConvolutionTop:PoolTop:XiDeconvTop:adapter --webtalk_flag SDSoC  --xp \"param:compiler.skipTimingCheckAndFrequencyScaling=1\" --xp "vivado_prop:run.impl_1.{STEPS.OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp \"vivado_prop:run.impl_1.STEPS.PHYS_OPT_DESIGN.IS_ENABLED=1\" --xp "vivado_prop:run.impl_1.{STEPS.PHYS_OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.ROUTE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp \"vivado_prop:run.synth_1.{STEPS.SYNTH_DESIGN.TCL.PRE}={D:\SourceCode\2018\AI\CHaiDNN\design\build/../conv/scripts/mcps.tcl}\" --xp \"vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.TCL.PRE}={D:\SourceCode\2018\AI\CHaiDNN\design\build/../conv/scripts/mcps.tcl}\" --xp \"param:compiler.deleteDefaultReportConfigs=false\" '
sds++ log file saved as D:/SourceCode/2018/AI/CHaiDNN/design/build/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed
sds++ completed at Sun Nov 11 09:48:01 +0800 2018

nkpavan, 2018-11-12:
@sheneuler ,

Are you trying to build CHaiDNN on Windows? If yes, please make sure that the path to mcps.tcl is as per Windows. The makefile provided works in Linux environment.

VishalX, 2018-12-06:
Closing. Please reopen if required.

closed 2018-12-06 by VishalX

DrChungAlbert, 2019-03-12:
I have the same problem on trying to build CHaiDNN version 2 on ZCU102 platform on SDSoC 2018.2.

===>The following messages were generated while  Compiling (top level synthesis checkpoint) dynamic region Log file: D:/Xilinx/SDx/workspace_2018/CHaiDNN/Release/_sds/p0/vivado/prj/prj.runs/synth_1/runme.log  :

runme.log contents:

*** Running vivado
    with args -log zcu102_wrapper.vds -m64 -product Vivado -mode batch -messageDb vivado.pb -notrace -source zcu102_wrapper.tcl

****** Vivado v2018.2 (64-bit)
  **** SW Build 2258646 on Thu Jun 14 20:03:12 MDT 2018
  **** IP Build 2256618 on Thu Jun 14 22:10:49 MDT 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source zcu102_wrapper.tcl -notrace
source <D:XilinxSDxworkspace_2018CHaiDNNsrcdesignconvscriptsmcps.tcl
ERROR: [runtcl-1] couldn't read file "<D:XilinxSDxworkspace_2018CHaiDNNsrcdesignconvscriptsmcps.tcl": no such file or directory
ERROR: [Common 17-39] 'send_msg_id' failed due to earlier errors.

    while executing
"send_msg_id runtcl-1 error "$_RESULT""
    invoked from within
"if {$src_rc} { 
  send_msg_id runtcl-1 error "$_RESULT"
  send_msg_id runtcl-2 error "sourcing script <D:XilinxSDxworkspace_2018CHaiDNNsrcdesignconvsc..."
    (file "zcu102_wrapper.tcl" line 49)
INFO: [Common 17-206] Exiting Vivado at Mon Mar 11 23:33:47 2019...

The correct path of my mcps.tcl is  "D:\Xilinx\SDx\workspace_2018\CHaiDNN\src\design\conv\scripts\mcps.tcl"

I had set SDS++ Linker with command
sds++  -xp param:compiler.skipTimingCheckAndFrequencyScaling=1 -xp "vivado_prop:run.impl_1.{STEPS.OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}"  -xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" -xp "vivado_prop:run.impl_1.STEPS.PHYS_OPT_DESIGN.IS_ENABLED=1" -xp "vivado_prop:run.impl_1.{STEPS.PHYS_OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" -xp "vivado_prop:run.impl_1.{STEPS.ROUTE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" -xp "vivado_prop:run.synth_1.{STEPS.SYNTH_DESIGN.TCL.PRE}={<**D:\Xilinx\SDx\workspace_2018\CHaiDNN\src\design\conv\scripts\mcps.tcl**}" -xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.TCL.PRE}={**D:\Xilinx\SDx\workspace_2018\CHaiDNN\src\design\conv\scripts\mcps.tcl**}" -Wno-unused-label

but the Vivado seems can't recognize the right path , and I got the log below:
ERROR: [runtcl-1] couldn't read file "<D:XilinxSDxworkspace_2018CHaiDNNsrcdesignconvscriptsmcps.tcl"

I want know how to set the path to let Vivado get the right path on windows ?