Xilinx/CHaiDNN #106: Build failed due to pthread issue

mattber, 2018-11-13:
Hi,

I'm trying to build CHaiDNN using SDSoC 2018.2 under Ubuntu. I first had the same problems as in #60, but after replacing ifdef with ifndef, that error went away. However, I get another error now:

Create data motion intermediate representation
/media/matthias/backup/sdsoc/workspaces/CHaiDNN/src/software/swkernels/xi_eltwiseadd_top.cpp:663:2: error: use of undeclared identifier 'pthread_exit'
        pthread_exit(NULL);
        ^
/media/matthias/backup/sdsoc/workspaces/CHaiDNN/src/software/swkernels/xi_eltwiseadd_top.cpp:678:2: error: unknown type name 'pthread_t'
        pthread_t eltwaddThread[NUM_ELTW_THREADS];
        ^
/media/matthias/backup/sdsoc/workspaces/CHaiDNN/src/software/swkernels/xi_eltwiseadd_top.cpp:714:9: warning: unused variable 'eltwaddRet' [-Wunused-variable]
                                int eltwaddRet = pthread_join(eltwaddThread[th], NULL);
                                    ^
ERROR: [SdsCompiler 83-5005] clang exited with non-zero code processing /media/matthias/backup/sdsoc/workspaces/CHaiDNN/src/software/swkernels/xi_eltwiseadd_top.cpp
sds++ log file saved as /media/matthias/backup/sdsoc/workspaces/CHaiDNN/Release/_sds/reports/sds_xi_eltwiseadd_top.log
ERROR: [SdsCompiler 83-5004] Build failed

src/software/swkernels/subdir.mk:39: recipe for target 'src/software/swkernels/xi_eltwiseadd_top.o' failed
make: *** [src/software/swkernels/xi_eltwiseadd_top.o] Error 1

15:47:15 Build Finished (took 1s.824ms)

Seems like he cannot find pthreads. Any idea?

anithabarri, 2018-11-20:
Hi mattber,

Are you using makefile setup or GUI? If GUI then make sure that you have added all the flags/symbols in build settings properly.

Thanks
Anitha

VishalX, 2018-12-06:
Closing. Please reopen if required.

closed 2018-12-06 by VishalX

TitechTraj, 2020-10-22:
I am trying to test ChiDNN on a custom platform and building it following the GUI instructions on https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md
I get the same error about pthread not declared.
I followed the instructions correctly and added all the flags and symbols (pthread is not mentioned though).
I tried to add pthread in the Libraries under SDS++ Linker, but still, the compiler can't find it.
I am not very familiar with SDx and the eclipse environment. Where exactly should I link the pthread library?

Thank you