Xilinx/CHaiDNN #107: What's the name of DepthwiseConvolutionLayer in the prototxt?

HelloSongRen, 2018-11-16:
Hi, does CHaiDNN support Depthwise Convolution Layer? What's the name of Depthwise Convolution Layer in the prototxt?
Thanks!

anilmartha, 2018-11-19:
Hi @HelloSongRen,

CHaiDNN supports Depthwise Convolution Layer. Caffe uses "Convolution" as the layer type for Depthwise Convolution in the prototxt. We consider a convolution as Depthwise Convolution if its number of outputs and groups are same.

HelloSongRen, 2018-11-19:
Great! Thank you~ @anilmartha

cuongdv1, 2018-11-28:
@anilmartha 
I tried to run MobilenetSSD on CHaiDNN and found that all depthwise convolution layers were removed in the function "void XGraphOpt::opt_separable_conv(XGraph *xgraph_opt)". You can check them in the attachment files as below:
[image]
[image]

I understanding that depth-wise conv of MobilenetSSD was not implemented on CHaiDNN?
could you tell me know how to modify CHaiDNN to keep depthwise conv of MobilenetSSD? I am investigating the source code of CHaiDNN, but it's difficult to understanding

Thank you very much.

VishalX, 2018-12-06:
@cuongdv1, 

#117
With the upcoming release, we are going to support MobileNet-SSD. 

Thanks!

closed 2018-12-06 by VishalX

mazhongzhong, 2019-06-14:
@VishalX,
Do you mean the ChaiDNN doesn't support Depthwise Convolution Layer for the time being? But why anilmartha said the ChaiDNN does support Depthwise Convolution Layer？And the document (https://github.com/Xilinx/CHaiDNN/blob/master/docs/SUPPORTED_LAYERS.md) says so as well.
If the ChaiDNN does support the Depthwise Convolution Layer, why it can not support MobileNet-SSD?