Xilinx/CHaiDNN #113: Problem of generating libxlnxdnn.so

lishen565, 2018-11-21:
Hi all, did anybody have generated the libxlnxdnn.so successfully?As the Xilinx SDx couldn't generate the libxlnxdnn.so file,  I have tried  to get a libxlnxdnn.so by using the makefile in design/build/, but something goes wrong as below:

make libxlnxdnn.so 
/opt/Xilinx/SDx/2017.4/bin/sds++ -sds-pf zcu102  -dmclkid 2  -xp param:compiler.skipTimingCheckAndFrequencyScaling=1 -xp "vivado_prop:run.impl_1.{STEPS.OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" -xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" -xp "vivado_prop:run.impl_1.STEPS.PHYS_OPT_DESIGN.IS_ENABLED=1" -xp "vivado_prop:run.impl_1.{STEPS.PHYS_OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" -xp "vivado_prop:run.impl_1.{STEPS.ROUTE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" -Wall -O3 -c -I. -I../conv/include -I../conv/src -I../pool/include -I../pool/src -I../deconv/include -I../deconv/src -D__SDSOC=1 -Wno-unused-label -D__SYNTHESIS__=1 -DXI_DIET_CHAI_Z=0 -DXI_DIET_CHAI_ZUPLUS=0 -D__CONV_ENABLE__=1 -D__POOL_ENABLE__=1 -D__DECONV_ENABLE__=1 -fPIC ../wrapper/dnn_wrapper.cpp -o dnn_wrapper.o
Platform system configuration option -sds-sys-config was not specified, searching for a configuration that uses the specified OS linux
Using system configuration -sds-sys-config a53_linux
Create data motion intermediate representation
/usr/lib/gcc-cross/aarch64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/cmath:41:10: fatal error: 'bits/c++config.h' file not found
#include <bits/c++config.h>
         ^
ERROR: [SdsCompiler 83-5005] clang exited with non-zero code processing /home/lishen/CHaiDNN-v2-master/CHaiDNN-master-v2/design/wrapper/dnn_wrapper.cpp
sds++ log file saved as /home/lishen/CHaiDNN-v2-master/CHaiDNN-master-v2/design/build/_sds/reports/sds_dnn_wrapper.log
ERROR: [SdsCompiler 83-5004] Build failed

Makefile:141: recipe for target 'dnn_wrapper.o' failed
make: *** [dnn_wrapper.o] Error 1

Can you help me solve this problem? Or is there any way to generate the libxlnxdnn.so in SDx?
Thanks a lot.

VishalX, 2018-11-21:
Hi @lishen565 

CHaiDNN-v2 is not supported on SDx-2017.4. Please use SDx-2018.2 to build the design.

lishen565, 2018-11-21:
Hi, in fact SDx-2017.4 can also compile ChaiDNN-v2 and I had got the result *.elf from it. I've also tried 2018.2 and "make libxlnxdnn.so" in console, but the error info is the same.

VishalX, 2018-12-06:
Hi @lishen565 

We have tried to reproduce the issue but we did not see the error you're having. Could you please clean your previous build and rebuild it again?

Thanks!

closed 2019-01-09 by VishalX