Xilinx/CHaiDNN #118: what's the meaning of the argument "--input_scale" for XportDNN?

mazhongzhong, 2018-11-29:
the argument "--raw_scale", is used for scale the input values. What's the argument "--input_scale" for? There is no explanation in the doc for this argument. What's the difference between "--raw_scale" and "--input_scale"?

anilmartha, 2018-11-30:
HI @mazhongzhong 
•	Raw-scale is a scale of your input data.  If your input data has values between 0 and 255, then raw_scale set to 255. If your data has values between 0 and 1, then raw_scale should be set to 1.
•	Input scale is a scaling factor. This is done after subtraction of mean and other preprocessing only if you have used scaling factor in training. If you have not, then input scale should be 1.

DanielsJusts, 2018-11-30:
Hello, @mazhongzhong!

I want to add to @anilmartha's answer:
- `--raw_scale` is used to scale raw data for preprocessing. To be exact - to scale the data for subtraction of the mean from input data. Basically, when the image is read, it's values are in range `[0,1]`. You can scale them to subtract the mean.
- `--input_scale` is used to scale the processed data before inputting them in the network.

I figured this out by checking out `caffe.io.Transformer` in Caffe's GitHub. You can check it out [here](https://github.com/BVLC/caffe/blob/99bd99795dcdf0b1d3086a8d67ab1782a8a08383/python/caffe/io.py#L122).

closed 2018-12-06 by VishalX