Xilinx/CHaiDNN #120: Quantization of MobileNet-SSD fails

salcanmor, 2018-12-05:
Hi guys, I've read #53 and #42 but I cannot quantize MobileNet-SSD.

This is what I've done:
1- Download prototxt: https://github.com/chuanqi305/MobileNet-SSD/blob/master/deploy.prototxt
2- Download caffe model: https://drive.google.com/open?id=0B3gersZ2cHIxRm5PMWRoTkdHdHc
3- Run XportDNN.pyc:

```
python XportDNN.pyc --quant_type "Xilinx" \
--deploy_model ./mobilenet/deploy.prototxt \
--weights ./mobilenet/MobileNetSSD_deploy.caffemodel \
--quantized_deploy_model ./mobilenet/quantized_deploy.prototxt \
--calibration_directory ./xilinx_quant_example_model/sample_calibration_dataset/ --calibration_size 10 \
--bitwidths 6,6,6 --dims 3,224,224 --transpose 2,0,1 \
--channel_swap 2,1,0 --raw_scale 255.0 \
--mean_value 127.5 --input_scale 0.007843
```
Please find attached both prototxt and caffemodel.

[MobileNet-SSD.zip](https://github.com/Xilinx/CHaiDNN/files/2648003/MobileNet-SSD.zip)

What am I doing wrong?

Thanks in advance!

cuongdv1, 2018-12-05:
@salcanmor 
Have you modified the first layer in prototxt as below?
==
name: "MobileNet-SSD"
layer {
  name: "data"
  type: "Input"
  top: "data"
  input_param {
    shape {
      dim: 1
      dim: 3
      dim: 300
      dim: 300
    }
  }
}
==

salcanmor, 2018-12-05:
> @salcanmor
> Have you modified the first layer in prototxt as below?
> 
> name: "MobileNet-SSD"
> layer {
> name: "data"
> type: "Input"
> top: "data"
> input_param {
> shape {
> dim: 1
> dim: 3
> dim: 300
> dim: 300
> }
> }
> }

I've tried that with no luck. See prototxt attached.
Did you manage to make the quantization of MobileNet-SSD work?

[MobileNet_deploy_orig.prototxt.zip](https://github.com/Xilinx/CHaiDNN/files/2648662/MobileNet_deploy_orig.prototxt.zip)

cuongdv1, 2018-12-06:
@salcanmor I made the quantization of Mobilenet-SSD and can run MobilenetSSD on CHaiDNN but the accuracy is too bad. I think depthwise conv was not supported, I am waiting for the help

salcanmor, 2018-12-06:
> @salcanmor I made the quantization of Mobilenet-SSD and can run MobilenetSSD on CHaiDNN but the accuracy is too bad. I think depthwise conv was not supported, I am waiting for the help

Can you please share the model and prototxt you used?
Thanks in advance!

cuongdv1, 2018-12-06:
Prototxt and caffemodel I cloned from here https://github.com/chuanqi305/MobileNet-SSD

salcanmor, 2018-12-07:
> Prototxt and caffemodel I cloned from here https://github.com/chuanqi305/MobileNet-SSD

If we are using the same Prototxt and caffemodel, why do I get an error and you not ?

cuongdv1, 2018-12-10:
Can you try this?

python XportDNN.pyc --quant_type "Xilinx" 
--deploy_model ./models/MobilenetSSD_300_deploy.prototxt 
--weights ./models/MobilenetSSD_300_deploy.caffemodel 
--quantized_deploy_model ./models/MobilenetSSD_300_quantized_deploy.prototxt 
--calibration_directory ./data/VOC0712 --calibration_size 32 
--bitwidths 8,8,8 --dims 3,300,300 --transpose 2,0,1 
--channel_swap 2,1,0 --raw_scale 255.0 
--mean_value 127.5,127.5,127.5 --input_scale 0.007843

salcanmor, 2018-12-10:
> Can you try this?
>
> python XportDNN.pyc --quant_type "Xilinx"
> --deploy_model ./models/MobilenetSSD_300_deploy.prototxt
> --weights ./models/MobilenetSSD_300_deploy.caffemodel
> --quantized_deploy_model [...]

I'm gettting the same error:
```
F1210 09:14:50.208392 27097 net.cpp:813] Check failed: target_blobs.size() == source_layer.blobs_size() (1 vs. 2) Incompatible number of blobs for layer conv1/dw
*** Check failure stack trace: ***
Abortado (`core' generado)
```

HelloSongRen, 2018-12-19:
> @salcanmor I made the quantization of Mobilenet-SSD and can run MobilenetSSD on CHaiDNN but the accuracy is too bad. I think depthwise conv was not supported, I am waiting for the help

Hi, can you please share the FPS of MobilenetSSD running on ZCU102?
Thanks a lot~

HelloSongRen, 2018-12-20:
@cuongdv1 ,  
Hi, can you please share the FPS of MobilenetSSD running on ZCU102?
Thanks a lot~

cuongdv1, 2018-12-20:
30fps, but the accuracy is too bad. i think it did not run full network

HelloSongRen, 2018-12-22:
Thank you!
Maybe the bad accuracy is caused by the Xilinx quantization, it is not good enough to quantize the light models like mobilenet......

averr5, 2019-05-01:
@salcanmor I am trying to run the MobilenetSSD. 

python XportDNN.pyc --quant_type "Xilinx" \
--deploy_model ./models/MobileNet_deploy_orig.prototxt \
--weights ./models/MobileNetSSD_deploy.caffemodel \
--quantized_deploy_mode ./models/MobilenetSSD_300_quantized_deploy.prototxt \
--calibration_directory ./xilinx_quant_example_model/sample_calibration_dataset \
--bitwidths 8,8,8 --dims 3,300,300 --transpose 2,0,1 \
--channel_swap 2,1,0 --raw_scale 255.0 \
--mean_value 127.5,127.5,127.5 --input_scale 0.007843

I am running into the same error as yours. 

I0501 22:42:27.699175  8893 net.cpp:807] Ignoring source layer data_input_0_split
F0501 22:42:27.699188  8893 net.cpp:813] Check failed: target_blobs.size() == source_layer.blobs_size() (1 vs. 2) Incompatible number of blobs for layer conv0
*** Check failure stack trace: ***
Aborted (core dumped)

Do you have any idea why ? /home/eva/Downloads/models.zip
PFA the caffemodel and prototxt file I have used.

averr5, 2019-05-02:
@cuongdv1 Did you use deploy.prototxt and mobilenet_iter_73000.caffemodel as inputs ?