Xilinx/CHaiDNN #124: ZCU102 runs tiny-yolo-v2 failed

dongkesi, 2018-12-06:
Hello,

I get the YOLO-V2 model from this website: https://github.com/tsingjinyun/caffe-yolov2.
Then use the following script to quantize the model, everthing looks OK.
```
python XportDNN.pyc --quant_type "Xilinx" \
--deploy_model ./xilinx_quant_example_model/yolo-v2/tiny_yolo_voc_deploy.prototxt \
--weights ./xilinx_quant_example_model/yolo-v2/tiny_yolo_voc.caffemodel \
--quantized_deploy_model ./xilinx_quant_example_model/yolo-v2/tiny_yolo_voc_quantized_deploy.prototxt \
--calibration_directory ./xilinx_quant_example_model/voc_calibration \
--calibration_size 70 \
--bitwidths 8,8,8 --dims 3,416,416 --transpose 2,0,1 \
--channel_swap 2,1,0 --raw_scale 255.0 \
--mean_value 127.5,127.5,127.5 --input_scale 0.007843
```
After that I use the ssd_ex.cpp, change a little code to load this model on zcu102, get the following output. The out_1_0.txt and out_0_0.txt is empty.
```
[WARNING] Precision Parameters are not specified correctly for some of the layers.
So CHaiDNN assumes default values for those parameters. It might affect the accuracy of result.
For accurate results, please provide the correct parameters for following layers:
'conv1_bn'      'conv1_scale'   'conv2_bn'      'conv2_scale'   'conv3_bn'
'conv3_scale'   'conv4_bn'      'conv4_scale'   'conv5_bn'      'conv5_scale'
'conv6_bn'      'conv6_scale'   'conv7_bn'      'conv7_scale'   'conv8_bn'
'conv8_scale'
[INFOx] Graph generated
[INFOx] Generating JobQueue
 CRITICAL WARNING: A drop in accuracy may be Observed
Following producer and consumer layes threshold values are not matching
Producer layer name:conv1
Consumer layer name:pool1
 CRITICAL WARNING: A drop in accuracy may be Observed
Following producer and consumer layes threshold values are not matching
Producer layer name:conv2
Consumer layer name:pool2
 CRITICAL WARNING: A drop in accuracy may be Observed
Following producer and consumer layes threshold values are not matching
Producer layer name:conv3
Consumer layer name:pool3
 CRITICAL WARNING: A drop in accuracy may be Observed
Following producer and consumer layes threshold values are not matching
Producer layer name:conv4
Consumer layer name:pool4
 CRITICAL WARNING: A drop in accuracy may be Observed
Following producer and consumer layes threshold values are not matching
Producer layer name:conv5
Consumer layer name:pool5
 CRITICAL WARNING: A drop in accuracy may be Observed
Following producer and consumer layes threshold values are not matching
Producer layer name:conv6
Consumer layer name:pool6
 CRITICAL WARNING: A drop in accuracy may be Observed
Following producer and consumer layes threshold values are not matching
Producer layer name:conv7
Consumer layer name:conv8
 CRITICAL WARNING: A drop in accuracy may be Observed
Following producer and consumer layes threshold values are not matching
Producer layer name:conv8
Consumer layer name:conv9
[INFOx] JobQueue generated
[INFOx] Creating Memory

```
It seems that quantification is not successful. Does anyone know which step has a problem?

Thanks,
Kedong
[tiny_yolo_voc_quantized_deploy.prototxt.txt](https://github.com/Xilinx/CHaiDNN/files/2652328/tiny_yolo_voc_quantized_deploy.prototxt.txt)

anilmartha, 2018-12-06:
Hi @Dongkesi,

The model you are using seems to have batch norm and scale layers. Currently, XportDNN does not support batchnorm and scale layers individually. You can fuse batchnorm and scale layers to convolution and create protoxt and caffe model. Then, run XportDNN with fused prototxt and fused caffemodel.

dongkesi, 2018-12-06:
Hi @anilmartha  Thanks your information, I use this tool https://github.com/HolmesShuan/Caffe-Computation-Graph-Optimization, fuse BN and scale to conv, the warning go away, but the out_1_0.txt and out_1_0.txt are still empty.  I tried kinds of images from VOC.

anilmartha, 2018-12-06:
Hi @dongkesi,
Missed to add the information in the previous comment.
CHaiDNN does not have support for ReLU with negative slope (Leaky ReLU). You could retrain your model with regular relu and try use it on CHaiDNN.

dongkesi, 2018-12-07:
Hi @anilmartha Thanks, I just understand caffe relu layer has negative_slope param means leaky relu.

closed 2018-12-11 by anilmartha

JIAWENYANG, 2019-01-16:
@dongkesi ,does yolo finally works successful?

tmxklqy, 2019-01-18:
@dongkesi ,does yolo finally works successful?

dongkesi, 2019-02-20:
@JIAWENYANG @tmxklqy I just modified the activation function, retrained the network, and integrated bachnorm and scale, but after deployment there was always no output, and then it was temporarily stopped because of other work.

Sunyouteng, 2019-05-13:
> 
> 
> @JIAWENYANG @tmxklqy I just modified the activation function, retrained the network, and integrated bachnorm and scale, but after deployment there was always no output, and then it was temporarily stopped because of other work.

hello， can you share your retrained network parameter with leaky Relu, including the weights file and cfg file. what is the mAP? I want to continue this project with your work.

dongkesi, 2019-06-09:
I lost these files. You can follow [these steps](https://github.com/dongkesi/Autonomous-Full-Stack-Learning/blob/a540ef826016c1b635babca34cd495eb13955abf/Xilinx/Notes/CHaiDNN_tiny-yolo-v3.md) to trainning your weights.

Sunyouteng, 2019-06-10:
https://github.com/Sunyouteng/yolov2-cfg-relu
I retrained the network with relu and it's weightsfile can be downloaded.

JIAWENYANG, 2019-06-10:
@Sunyouteng can you show more details with us?step bys tep