Xilinx/CHaiDNN #132: How to run ChaiDNN in emulation mode with SDx?

sjtu-nobody, 2018-12-29:
Hi, I have successfully build the CHaiDNN under a SDx project with the instructions, but I can't build the project for emulation in the same settings with hardware integration failure, the vivado log prints that vivado couldn't find some modules, for example: 
ERROR: [VRFC 10-2063] Module <axi_register_slice_v2_1_17_axi_register_slice> not found while processing module instance <inst> [/home/le8888e/workplace/ChaiDNN/Debug/_sds/p0/vivado/prj/prj.ip_user_files/bd/zcu102/ip/zcu102_m00_regslice_0/sim/zcu102_m00_regslice_0.v:186].
ERROR: [VRFC 10-2063] Module <zcu102_xbar_0> not found while processing module instance <xbar> [/home/le8888e/workplace/ChaiDNN/Debug/_sds/p0/vivado/prj/prj.ip_user_files/bd/zcu102/sim/zcu102.v:22836].
So, any suggestions about this problem? Thanks a lot.

WenyeLiu, 2019-03-19:
Hi, 
I have the same question. The CHaiDNN can be built successfully with Makefile flow. When I try to build it in emulation mode in SDx GUI, I meet some errors.

My error is not as yours @sjtu-nobody, it shows "src/design/build/_sds/.llvm/cf_stub.h:6:10: fatal error: 'portinfo.h' file not found". 

I compare the files with the Makefile flow and found that portinfo.h file is located under the vivado porject. I guess it might due to no hardware implementation in emulation mode. However, I have no idea how to fixed it.

averr5, 2019-04-04:
@WenyeLiu After building with Makefile, how did you proceed (steps) with building in emulation mode in SDx GUI ?

WenyeLiu, 2019-04-04:
@averr5 You don't need to build with Makefile if you use the SDx GUI. To build in emulation mode just follow the [instruction](https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md). And the only change you need to do is set the target to emulation mode, refer to [lab8](https://github.com/Xilinx/SDSoC-Tutorials/blob/master/getting-started-tutorial/lab-8-emulation.md)

averr5, 2019-04-28:
@WenyeLiu I am not able to build CHaiDNN using SDx GUI. Were you able to build it with SDx GUI? I was only able to build it using Makefile. Any idea on how to proceed with emulation?