Xilinx/CHaiDNN #134: CHaiDNN normalization input

eerobert, 2019-01-03:
I have a network trained with pixel/255.0 as normalized input

however, i cannot get the correct results.
i wonder do chaidnn accept input of pixels/255.0 as input layer?
any examples?

eerobert, 2019-01-03:
i notice that the xiInputRead will call only loadMeanSubtractedDatafromBuffptr
and there are no other loadNormalizedInputDatafromBuffptr(inp_mode == 1)
or loadPixelMeanSubtractedDatafromBuffptr (inp_mode == 2)

So does it mean it do not support normalized input with pixel/255.0?

NeutrinoXY, 2019-01-07:
I suggest you to create your own input layer if the ones provided don't fit your needs, it's simple.

eerobert, 2019-01-08:
so are you suggesting to modify the xiInputRead function?
can you elaborate how simple is to add a new xiInputRead function? thanks

eerobert, 2019-01-29:
still stuck in modifying xiInputRead function.
i wish to know what is the data type expected for such function?

eerobert, 2019-02-04:
anyone can help?

eerobert, 2019-02-27:
Hi, since the pixel/255 will result in input value (-1,1], what is the expected representation of such value?
in mean-subtracted case, it is rather understandable. the example use truncated input (as in Trim2FP_Offline).
However in the case of pixel/255, the Trim2FixedPoint will makes every input to 0