Xilinx/CHaiDNN #136: The official example and inference run is explained for w/ FC CNN, How to test the w/o FC performance?

linpeng9527, 2019-01-07:
Take AlexNet as example, the official AlexNet example is 9.98 images/s for w/ FC, but how to test the w/o FC 217 image/s ?

NeutrinoXY, 2019-01-07:
If I remember well, you can use the getPerfInfo function defined in software/xi_interface.cpp.

linpeng9527, 2019-01-09:
> If I remember well, you can use the getPerfInfo function defined in software/xi_interface.cpp.

Thanks for your reply, I am trying, can you speak more clearly?

VishalX, 2019-01-10:
@linpeng9527,

To check the performance of AlexNet w/o FC layers, the simplest way is to remove the FC layers from your prototxt and don't use graph splitting.

linpeng9527, 2019-01-16:
> If I remember well, you can use the getPerfInfo function defined in software/xi_interface.cpp.

@NeutrinoXY I got error when I simply removed  FC layers from prototxt. I will be appreciate it if you explain testing the "w/o FC performance" more clearly.

VishalX, 2019-01-16:
@linpeng9527 

If you see the [alexnet_ex.cpp](https://github.com/Xilinx/CHaiDNN/blob/master/software/example/alexnet_ex.cpp),

```
	//# start/end layer in the graph1
	string start_layer_graph1 = "";
	string end_layer_graph1   = "pool5";

	//# start/end layer in the graph2
	string start_layer_graph2 = "fc6";
	string end_layer_graph2   = "";

	//# Flag indicate layer1 or not
	//# 1 - if image data is the for first layer input otherwise 0
	bool is_first_layer_g1 = 1;
	bool is_first_layer_g2 = 0;

	//# Number of images to process
	//# Supported batch size is 2
	int numImg_to_process = 2;

	//# Struct which holds the input/output layer info
	io_layer_info io_layer_info_ptr1;
	io_layer_info io_layer_info_ptr2;

	//####### Init call for graph1
	void *chaihandle1 = xiInit(dirpath, prototxt, caffemodel, &io_layer_info_ptr1,
			numImg_to_process, is_first_layer_g1, start_layer_graph1, end_layer_graph1);

	//####### Init call for graph2
	void *chaihandle2 = xiInit(dirpath, prototxt, caffemodel, &io_layer_info_ptr2,
			numImg_to_process, is_first_layer_g2, start_layer_graph2, end_layer_graph2);
```
The network is splitted into 2 graphs (graph1 & graph2). Now, as you have removed the FC layers, you don't require graph2 at all and only need to initialize graph1 (`chaihandle1`).

See the [ssd_ex.cpp](https://github.com/Xilinx/CHaiDNN/blob/master/software/example/ssd_ex.cpp) which doesn't use graph splitting and make the necessary changes in your example file.

linpeng9527, 2019-01-17:
> @linpeng9527
>
> If you see the [alexnet_ex.cpp](https://github.com/Xilinx/CHaiDNN/blob/master/software/example/alexnet_ex.cpp),
>
> ```
> 	//# start/end layer in the graph1
> 	string start_layer_graph1 = [...]

@VishalX  Thanks for your advice. I changed the alexnet_ex.cpp as you said, but the performance is only 90 images/s. There is certain gap compared with offocial 217 images/s. Do you know the potential cause? The performance is as follows
[image]

VishalX, 2019-01-17:
@linpeng9527 

Can you share the design config, device, and frequency details?

linpeng9527, 2019-01-17:
> @linpeng9527
> 
> Can you share the design config, device, and frequency details?

@VishalX  Thanks for your reply. I build the project in SDx 2018.2 followed by official "Creating SDx GUI Project".
The device is KCU104 evalulation board. The frequency is 200MHz, The configure in “xi_conv_config.h” is as follows, so is "hw_settings.h".
1. 
//**** URAM ENABLE FLAGS FOR THE BUFFERS
#define XI_BIAS_URAM_EN 	0
#define XI_WTS_URAM_EN 	1
#define XI_ISTG_URAM_EN 	1
#define XI_OSTG_URAM_EN 	1
#define XI_FEED_URAM_EN 	0
#define XI_SCALE_URAM_EN 	0

//**** CONFIGURABLE MACROS
#if XI_DIET_CHAI_Z || XI_DIET_CHAI_ZUPLUS
#define XI_KER_PROC       	16
#define XI_PIX_PROC       	16
#define XI_ISTAGEBUFF_DEPTH 1024
#define XI_OSTAGEBUFF_DEPTH 1024
#define XI_WEIGHTBUFF_DEPTH 1024
#else
#define XI_KER_PROC       	16
#define XI_PIX_PROC       	16
#define XI_ISTAGEBUFF_DEPTH 8192
#define XI_OSTAGEBUFF_DEPTH 2048
#define XI_WEIGHTBUFF_DEPTH 2048	
#endif
2.
 POOL_ENABLE and DECONV_ENABLE are set to 0.

The w/ FC performance for AlexNet is 9.9 images/s，The w/o FC performance for AlexNet is 98 images/s.
Have I omitted anything?

VishalX, 2019-01-17:
@linpeng9527 

Set below environment variable,
```
export OPENBLAS_NUM_THREADS=3
```
Run
```
./<executable>.elf Xilinx 8 images/camel.jpg images/goldfish.JPEG
```

linpeng9527, 2019-01-17:
@VishalX 
The w/ FC performance for alexnet is 9 images/s and w/o FC performance is 89 images/s. It is all tested in the configure OPENBLAS_NUM_THREADS=3. When i set OPENBLAS_NUM_THREADS=2, the w/ FC is 6 images/s while the w/o FC is also 89 images/s. There is little performance difference in w/o FC

VishalX, 2019-01-17:
@linpeng9527 

The performance numbers quoted [here](https://github.com/Xilinx/CHaiDNN/blob/master/docs/PERFORMANCE_SNAPSHOT.md) for zu7 device are at 300/600 MHz. You are running at 200 MHz. This is one definite reason for the lower performance for your build. 

Could you also share the timing reports (mentioned below), so that we can confirm?
```
zcu104_wrapper_timing_summary_routed.rpt
zcu104_wrapper_utilization_placed.rpt
```
You can find them inside
```
_sds/p0/vivado/prj/prj.runs/impl_1 
```

linpeng9527, 2019-01-17:
> @linpeng9527
>
> The performance numbers quoted [here](https://github.com/Xilinx/CHaiDNN/blob/master/docs/PERFORMANCE_SNAPSHOT.md) for zu7 device are at 300/600 MHz. You are running at 200 MHz. This is [...]

@VishalX Thanks for your reply. The timing reports you mentioned is as follows.
[zcu104_wrapper_timing_summary_routed.rpt.tar.gz](https://github.com/Xilinx/CHaiDNN/files/2767385/zcu104_wrapper_timing_summary_routed.rpt.tar.gz)
[zcu104_wrapper_utilization_placed.rpt.tar.gz](https://github.com/Xilinx/CHaiDNN/files/2767386/zcu104_wrapper_utilization_placed.rpt.tar.gz)

I will try to improve the frequency. I always don't understand why there is two frequency 300MHz/600MHz, what does it mean, respectively?

linpeng9527, 2019-01-17:
> This is one definite reason for the lower performance for your build.

@VishalX  Hi, I will try to improve the frequency, but I don't think the frequency is the main reason. Because my 200MHz w/ FC performance 9 images/s is very similar to the official 300/600MHz 9.99 images/s. I am verifying this guess.

VishalX, 2019-01-17:
@linpeng9527 

By looking at your timing report, it seems you built the design at **100/200 MHz**, not at 200/400 MHz.

There are two frequencies because the compute DSPs are running at twice the frequency as that of the rest of the logic.

To build for 200/400 or 300/600 MHz, change the below in SDS++Compiler settings,

sds++ -sds-hw XiConvolutionTop  <path to design>/conv/src/xi_convolution_top.cpp -clkid **CLKID** -hls-tcl  <path to design>/conv/scripts/config_core.tcl -sds-end

Where, 
```
CLKID=3 for 200/400 MHz
CLKID=4 for 300/600 MHz
```
These are applicable for stock platforms. Also, make sure to set the Datamotion Clock Frequency accordingly in SDx GUI.

VishalX, 2019-01-17:
w/ FC performance will be similar for different frequencies because FC layers run on PS (ARM) and not on  PL. As FC layers' latency is very high, the overall impact on performance will be very less at lower frequencies.

averr5, 2019-05-01:
@NeutrinoXY I am running AlexNet on the  board and got performance in images/s. Now how do i use getPerfInfo function?