Xilinx/CHaiDNN #137: Questions about using XportDNN

comeonwmz, 2019-01-08:
I was trying using XportDNN to convert my prototxt of yolov3 to quantization mode. But there is an error when i run the script like this:
[image]
However, i have added the upsample layer to my caffe. So i wonder whether the XportDNN can convert the layer added by users or not. And i also want to know how you handle this issue (about the quantization params of Upsample layer & Yolo layer) cause i saw the Benchmarks at Performance_eval.md
[image]

xiemark, 2019-01-08:
CHaiDNN does not support leaky ReLU. So you need to retrain all the Yolo networks in darknet with normal ReLU. But if you retrain with normal ReLU, it is very difficult to converge.

comeonwmz, 2019-01-08:
@xiemark thanks for your hint, i'll figure out this later. Do you know the answer of what i mentioned about the XportDNN?

xiemark, 2019-01-08:
https://github.com/Xilinx/CHaiDNN/issues/124

JIAWENYANG, 2019-01-16:
@comeonwmz  I do the same work as you,how do you solve your question?how does yolov3 work successful???? @xiemark @comeonwmz

comeonwmz, 2019-02-25:
@JIAWENYANG  Not yet. If you make any progress, please tell me. Thanks