Xilinx/CHaiDNN #139: CHaiDNN layer supported Issues with Power layer

guru3110, 2019-01-10:
In the CHaiDNN documents tells that it support Power Layer but none of the implemented models is having PowerLayer support. This has been checked by i was trying to use the Googlenet model and our application specific (prototxt and caffemodel) were feeded into the model. Build the program is successful. But the when i run the application I am encounter the issue. The error is attached in this document.

guru3110, 2019-01-11:
> Hi @guru3110 <https://github.com/guru3110> , could you please share your
> prototxt here?

HI Abid,

abid-k, 2019-01-11:
Hi @guru3110,

Power Layer is supported only in the context of CReLU. Not as a stand-alone Power Layer.

guru3110, 2019-01-11:
Hi Abid,

Ok I understand it. So if we removed the power layer in our model because
the power layer in our model doesn't going to have big impact on the
inference. If we removed and check then it should work right. What it's
opining regarding this?

abid-k, 2019-01-11:
Hi @guru3110 

I can't tell that without seeing the entire network. 

But it is definitely worth a try. You can easily modify the network in prototxt and run it again.

guru3110, 2019-01-11:
ls Hi Abid,

I have tried like that and the output of that I am attaching in this email.
Please look into it and let me know. Is there a way to train the model for
CHaiDNN supported version.

-- 
*Be Happy,Be Enthusiastic, and Be Relax.*
With best regards,
Guruprasadh J P
Project-Intern
Ignitarium Technology Solutions Private Limited
+919486380752

Entered the main function
Unexpected element. Expected: idx1. Got: movi.
Failed to parse avi: index was not found
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
VIDEOIO ERROR: V4L: device ./outcpp.avi: Unable to query number of channels

(RegImg:2852): GStreamer-CRITICAL **: gst_query_set_position: assertion 'format == g_value_get_enum (gst_structure_id_get_value (s, GST_QUARK (FORMAT)))' failed
[INFOx] Network Path : models/GoogleNetWithoutLRN/6Bit
[PARSE] Parsing conv1
[PARSE] Parsing pool1
[PARSE] Parsing conv2
[PARSE] Parsing pool2
[PARSE] Parsing ip1
[PARSE] Parsing relu1
[PARSE] Parsing ip2
[PARSE] Parsing softmax
[IG001] Extracting conv1 weights ... 
[IG001] Extracting conv1 bias ... 
[IG001] Extracting conv2 weights ... 
[IG001] Extracting conv2 bias ... 
[IG001] Extracting ip1 weights ... 
[IG001] Extracting ip1 bias ... 

[WARNING] Precision Parameters are not specified correctly for some of the layers.
So xfDNN assumes default values for those parameters. It might affect the accuracy of result.
For accurate results, please provide the correct parameters for following layers: 
'conv1'	'conv2'	
[INFOx] Graph generated
[INFOx] Generating JobQueue
[INFOx] JobQueue generated
[INFOx] Creating Memory

[INFOx] Memory created
[INFOx] Network Path : models/GoogleNetWithoutLRN/6Bit

[INFOx] Init Start : This may take a while ...
[INFOx] Init Done
[PARSE] Parsing conv1
[PARSE] Parsing pool1
[PARSE] Parsing conv2
[PARSE] Parsing pool2
[PARSE] Parsing ip1
[PARSE] Parsing relu1
[PARSE] Parsing ip2
[PARSE] Parsing softmax
[IG001] Extracting ip2 weights ... 
[IG001] Extracting ip2 bias ... 

[WARNING] Precision Parameters are not specified correctly for some of the layers.
So xfDNN assumes default values for those parameters. It might affect the accuracy of result.
For accurate results, please provide the correct parameters for following layers: 
'conv1'	'conv2'	
[INFOx] Graph generated
[INFOx] Generating JobQueue
[INFOx] Packinfo fields are not matching
[INFOx] JobQueue generated
[INFOx] Creating Memory

[INFOx] Memory created
[INFOx] Network Path : models/GoogleNetWithoutLRN/6Bit

[INFOx] Init Start : This may take a while ...
[INFOx] Init Done
[ERROR] invalid resize params
[ERROR] Image Read fail
Seg fault @ 0x30
/run/media/mmcblk0p1/zcu104/lib/libxlnxdnn.so(+0x2a410)[0x7f9c1ae410]

guru3110, 2019-01-16:
Hey I am having error like "**Attempting to write axi-lite port that lacks a write status register**" when I build the googlenet working code in SDSoC. How to resolve this error. Please help me it's very urgent....

guru3110, 2019-02-01:
Hi Abid,

We have trained the dataset based on available googlenet.prototxt (80000 iteration we trained for 1000 images). But when I give the image as input for testing always it's printing same kind of accuracy as output. Classification for 2 items (either good or bad classification) class id 0 - good and class id 1 - bad score printing for class id 0 - 65% and class id 1 - 35%. This is output i am getting when i tried to run the inference based on the googlenet inference for our datasets. Why it's so like this??? Could you please help me in solving this issues.