Xilinx/CHaiDNN #141: Facing errors when following instructions to build using SDx GUI

NP95, 2019-01-17:
I am having a Ubuntu 18.04.1 LTS and downloaded SDSoC 2018.3 , I followed the instructions in this link 
https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md . But I am getting these errors while trying to build the project.

fatal error: 'google/protobuf/stubs/common.h' file not found
make: *** [sd_card] Error 1
make: *** [src/software/xtract/caffe.pb.o] Error 1
recipe for target 'src/software/xtract/caffe.pb.o' failed
SdsCompiler 83-5004: Build failed
SdsCompiler 83-5176: RTL template analysis exited with non-zero code processing /home/nishant/workspace/ChaiDNN_bringup/Release/__CONV_ENABLE__
/home/nishant/workspace/ChaiDNN_bringup/Release/__DSP48E2__
/home/nishant/workspace/ChaiDNN_bringup/Release/__SYNTHESIS__ /home/nishant/workspace/ChaiDNN_bringup/Release/__POOL_ENABLE__ /home/nishant/workspace/ChaiDNN_bringup/Release/__DECONV_ENABLE__ /home/nishant/workspace/ChaiDNN_bringup/src/software/xtract/caffe.pb.cc
TemplateExtract 83-3509: Failed to run clang frontend action of rtl_template_extract!

Can someone point me to what I am doing incorrectly that is generating these errors?

VishalX, 2019-01-21:
@NP95 

Check #57.

NP95, 2019-01-21:
Ok But I am still getting these errors

fatal error: 'google/protobuf/stubs/common.h' file not found
make: *** [sd_card] Error 1
make: *** [src/software/xtract/caffe.pb.o] Error 1
recipe for target 'src/software/xtract/caffe.pb.o' failed
SdsCompiler 83-5004: Build failed
SdsCompiler 83-5176: RTL template analysis exited with non-zero code processing /home/nishant/workspace/new/ChaiDNN_dummies/src/software/xtract/caffe.pb.cc
TemplateExtract 83-3509: Failed to run clang frontend action of rtl_template_extract!

VishalX, 2019-01-23:
Looks like include error. Make sure to provide correct include paths in settings.

NP95, 2019-01-25:
I moved to CentOS and followed the instructions, but I am getting errors related to gnu-aarch 64 compiler. I don't have it installed by default, but I believe it comes installed with SDSoc. Please look at the logs that I am attaching
'
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:41:9: error: unknown type name '__Int64x1_t'; did you mean '__int64_t'?
typedef __Int64x1_t int64x1_t;
        ^~~~~~~~~~~
        __int64_t
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/libc/usr/include/bits/types.h:43:25: note: '__int64_t' declared here
typedef signed long int __int64_t;
                        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:42:9: error: unknown type name '__Float16x4_t'
typedef __Float16x4_t float16x4_t;
        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:43:9: error: unknown type name '__Float32x2_t'
typedef __Float32x2_t float32x2_t;
        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:44:9: error: unknown type name '__Poly8x8_t'
typedef __Poly8x8_t poly8x8_t;
        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:45:9: error: unknown type name '__Poly16x4_t'
typedef __Poly16x4_t poly16x4_t;
        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:46:9: error: unknown type name '__Uint8x8_t'
typedef __Uint8x8_t uint8x8_t;
        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:47:9: error: unknown type name '__Uint16x4_t'
typedef __Uint16x4_t uint16x4_t;
        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:48:9: error: unknown type name '__Uint32x2_t'
typedef __Uint32x2_t uint32x2_t;
        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:49:9: error: unknown type name '__Float64x1_t'
typedef __Float64x1_t float64x1_t;
        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:50:9: error: unknown type name '__Uint64x1_t'
typedef __Uint64x1_t uint64x1_t;
        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:51:9: error: unknown type name '__Int8x16_t'; did you mean '__int16_t'?
typedef __Int8x16_t int8x16_t;
        ^~~~~~~~~~~
        __int16_t
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/libc/usr/include/bits/types.h:38:26: note: '__int16_t' declared here
typedef signed short int __int16_t;
                         ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:52:9: error: unknown type name '__Int16x8_t'
typedef __Int16x8_t int16x8_t;
        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:53:9: error: unknown type name '__Int32x4_t'; did you mean '__int32_t'?
typedef __Int32x4_t int32x4_t;
        ^~~~~~~~~~~
        __int32_t
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/libc/usr/include/bits/types.h:40:20: note: '__int32_t' declared here
typedef signed int __int32_t;
                   ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:54:9: error: unknown type name '__Int64x2_t'; did you mean '__int64_t'?
typedef __Int64x2_t int64x2_t;
        ^~~~~~~~~~~
        __int64_t
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/libc/usr/include/bits/types.h:43:25: note: '__int64_t' declared here
typedef signed long int __int64_t;
                        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:55:9: error: unknown type name '__Float16x8_t'
typedef __Float16x8_t float16x8_t;
        ^
/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:56:9: error: unknown type name '__Float32x4_t'
typedef __Float32x4_t float32x4_t; '

Even if I am trying to build using makefile in Ubuntu I am getting an error related to the compiler. Are there any changes I have to make with respect to SDSoc 2018.3 or gnu-aarch 64

NP95, 2019-01-30:
I have uploaded my Makefile and the platforms file. The procedure I did 

1. Clone the CHaiDNN repository
2. Go the the design/build folder and change the makefile according to the instructions here.
https://github.com/Xilinx/CHaiDNN  (under the Build from source instructions)

[zcu102.zip](https://github.com/Xilinx/CHaiDNN/files/2810691/zcu102.zip)

[Makefile.zip](https://github.com/Xilinx/CHaiDNN/files/2810685/Makefile.zip)

The error I am getting is 
`make
sds++ -sds-pf "/home/nishant/CHaiDNN/zcu102"  -dmclkid 1  -xp param:compiler.skipTimingCheckAndFrequencyScaling=1 -xp "vivado_prop:run.impl_1.{STEPS.OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" -xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" -xp "vivado_prop:run.impl_1.STEPS.PHYS_OPT_DESIGN.IS_ENABLED=1" -xp "vivado_prop:run.impl_1.{STEPS.PHYS_OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" -xp "vivado_prop:run.impl_1.{STEPS.ROUTE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" -Wall -O0 -c -I. -I../conv/include -I../conv/src -I../pool/include -I../pool/src -I../deconv/include -I../deconv/src -D__SDSOC=1 -Wno-unused-label -D__SYNTHESIS__=0 -DXI_DIET_CHAI_Z=0 -DXI_DIET_CHAI_ZUPLUS=0 -D__CONV_ENABLE__=1 -D__POOL_ENABLE__=1 -D__DECONV_ENABLE__=1 -fPIC ../wrapper/dnn_wrapper.cpp -o dnn_wrapper.o
Platform system configuration option -sds-sys-config was not specified, searching for a configuration that uses the specified OS linux
Using system configuration -sds-sys-config a53_linux
Analyzing source for RTL template usage
/tools/Xilinx/Vivado/2018.3/include/ap_int_base.h:265:19: error: use of undeclared identifier '__builtin_bit_select'
    bool is_neg = _AP_ROOT_op_get_bit(reg.V, BITS - 1);
                  ^
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:622:39: note: expanded from macro '_AP_ROOT_op_get_bit'
#define _AP_ROOT_op_get_bit(Val, Bit) _ssdm_op_get_bit(Val, Bit)
                                      ^
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:529:23: note: expanded from macro '_ssdm_op_get_bit'
    bool __Result__ = __builtin_bit_select((void*)(&__Val2__), Bit); \
                      ^
/tools/Xilinx/Vivado/2018.3/include/ap_int_base.h:268:13: error: use of undeclared identifier '__builtin_bit_part_select'
    exp.V = _AP_ROOT_op_get_range(reg.V, FLOAT_MAN, BITS - 2);
            ^
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:624:44: note: expanded from macro '_AP_ROOT_op_get_range'
#define _AP_ROOT_op_get_range(Val, Lo, Hi) _ssdm_op_get_range(Val, Lo, Hi)
                                           ^
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:547:5: note: expanded from macro '_ssdm_op_get_range'
    __builtin_bit_part_select((void*)(&__Result__), (void*)(&__Val2__), Lo, \
    ^
/tools/Xilinx/Vivado/2018.3/include/ap_int_base.h:272:13: error: use of undeclared identifier '__builtin_bit_part_select'
    man.V = _AP_ROOT_op_get_range(reg.V, 0, FLOAT_MAN - 1);
            ^
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:624:44: note: expanded from macro '_AP_ROOT_op_get_range'
#define _AP_ROOT_op_get_range(Val, Lo, Hi) _ssdm_op_get_range(Val, Lo, Hi)
                                           ^
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:547:5: note: expanded from macro '_ssdm_op_get_range'
    __builtin_bit_part_select((void*)(&__Result__), (void*)(&__Val2__), Lo, \
    ^
/tools/Xilinx/Vivado/2018.3/include/ap_int_base.h:277:13: error: use of undeclared identifier '__builtin_bit_part_set'
    man.V = _AP_ROOT_op_set_bit(man.V, FLOAT_MAN, 1);
            ^
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:623:45: note: expanded from macro '_AP_ROOT_op_set_bit'
#define _AP_ROOT_op_set_bit(Val, Bit, Repl) _ssdm_op_set_bit(Val, Bit, Repl)
                                            ^
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:538:5: note: expanded from macro '_ssdm_op_set_bit'
    __builtin_bit_part_set((void*)(&__Result__), (void*)(&__Val2__),   \
    ^
/tools/Xilinx/Vivado/2018.3/include/ap_int_base.h:313:19: error: use of undeclared identifier '__builtin_bit_select'
    bool is_neg = _AP_ROOT_op_get_bit(reg.V, BITS - 1);
                  ^
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:622:39: note: expanded from macro '_AP_ROOT_op_get_bit'
#define _AP_ROOT_op_get_bit(Val, Bit) _ssdm_op_get_bit(Val, Bit)
                                      ^
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:529:23: note: expanded from macro '_ssdm_op_get_bit'
    bool __Result__ = __builtin_bit_select((void*)(&__Val2__), Bit); \
                      ^
/tools/Xilinx/Vivado/2018.3/include/ap_int_base.h:316:13: error: use of undeclared identifier '__builtin_bit_part_select'
    exp.V = _AP_ROOT_op_get_range(reg.V, DOUBLE_MAN, BITS - 2);
            ^
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:624:44: note: expanded from macro '_AP_ROOT_op_get_range'
#define _AP_ROOT_op_get_range(Val, Lo, Hi) _ssdm_op_get_range(Val, Lo, Hi)
                                           ^
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:547:5: note: expanded from macro '_ssdm_op_get_range'
    __builtin_bit_part_select((void*)(&__Result__), (void*)(&__Val2__), Lo, \
    ^
[... 102 lines cut ...]
/tools/Xilinx/Vivado/2018.3/include/ap_common.h:623:45: note: expanded from macro '_AP_ROOT_op_set_bit'
#define _AP_ROOT_op_set_bit(Val, Bit, Repl) _ssdm_op_set_bit(Val, Bit, Repl)
                                            ^
[... cut ...]
ERROR: [TemplateExtract 83-3509] Failed to run clang frontend action of rtl_template_extract!
ERROR: [SdsCompiler 83-5176] RTL template analysis exited with non-zero code processing /home/nishant/CHaiDNN/design/wrapper/dnn_wrapper.cpp
sds++ log file saved as /home/nishant/CHaiDNN/design/build/_sds/reports/sds_dnn_wrapper.log
ERROR: [SdsCompiler 83-5004] Build failed
Makefile:143: recipe for target 'dnn_wrapper.o' failed
make: *** [dnn_wrapper.o] Error 1
`
Environment: Ubuntu 18.04 with kernel version 4.15.0-44-generic. SDSoc 2018.3

nkpavan, 2019-01-30:
@NP95 
As you are using SDSoc 2018.3, remove the __SYNTHESIS__ flag from Makefile and you need to include the stdio header in the pooling_layer_dp_2xio_top.cpp file.

NP95, 2019-01-31:
Closing this as I am able to build hardware with latest SDSoc. WIll open a new issue thread if running into issues with building software.

closed 2019-01-31 by NP95

averr5, 2019-04-08:
@nkpavan I am using SDSoC 2018.3. I have removed SYNTHESIS flag from Makefile also tried adding stdio header in pooling_layer_dp_top.cpp file. I am following steps from [https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md ](url), but i run into these errors.

Finished building: ../src/software/swkernels/xi_crop.cpp
 
Building file: ../src/software/swkernels/xi_eltwiseadd_top.cpp
Invoking: SDS++ Compiler
sds++ -sds-hw XiConvolutionTop  /home/eva/Workspace2/CHai/src/design/conv/src/xi_convolution_top.cpp -clkid 1 -hls-tcl  /home/eva/Workspace2/CHai/src/design/conv/scripts/config_core.tcl -sds-end -D__SDSOC -D__POOL_ENABLE__ -D__DECONV_ENABLE__ -D__CONV_ENABLE__ -D__DSP48E2__ -Wall -O0 -I"../src" -I/home/eva/Workspace2/CHai/src/design/pool/include -I/home/eva/Workspace2/CHai/src/design/pool/src -I/home/eva/Workspace2/CHai/src/design/deconv/include -I/home/eva/Workspace2/CHai/src/design/deconv/src -I/home/eva/Desktop/CHaiDNN/SD_Card/opencv/arm64/include -I/home/eva/Desktop/CHaiDNN/SD_Card/protobuf/arm64/include -I/home/eva/Desktop/CHaiDNN/SD_Card/cblas/arm64/include -I/home/eva/Workspace2/CHai/src/design/conv/include -I/home/eva/Workspace2/CHai/src/design/conv/src -c -fmessage-length=0 -MT"src/software/swkernels/xi_eltwiseadd_top.o" -MMD -MP -MF"src/software/swkernels/xi_eltwiseadd_top.d" -MT"src/software/swkernels/xi_eltwiseadd_top.o" -o "src/software/swkernels/xi_eltwiseadd_top.o" "../src/software/swkernels/xi_eltwiseadd_top.cpp" -sds-hw PoolTop pooling_layer_dp_2xio_top.cpp  -files ../src/design/pool/src/pool_dp_2xio.hpp  -clkid 1 -sds-end -sds-hw XiDeconvTop xi_deconv_top.cpp  -files ../src/design/deconv/src/xi_deconv.hpp  -clkid 1 -sds-end -sds-sys-config a53_linux -sds-proc a53_linux -sds-pf "/home/eva/Desktop/CHaiDNN/zcu102"
Analyzing source for RTL template usage
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:38:9: error: unknown type name '__Int8x8_t'
typedef __Int8x8_t int8x8_t;
        ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:39:9: error: unknown type name '__Int16x4_t'
typedef __Int16x4_t int16x4_t;
        ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:40:9: error: unknown type name '__Int32x2_t'; did you mean '__int32_t'?
typedef __Int32x2_t int32x2_t;
        ^~~~~~~~~~~
        __int32_t
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/libc/usr/include/bits/types.h:40:20: note: '__int32_t' declared here
typedef signed int __int32_t;
                   ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:41:9: error: unknown type name '__Int64x1_t'; did you mean '__int64_t'?
typedef __Int64x1_t int64x1_t;
        ^~~~~~~~~~~
        __int64_t
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/libc/usr/include/bits/types.h:43:25: note: '__int64_t' declared here
typedef signed long int __int64_t;
                        ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:42:9: error: unknown type name '__Float16x4_t'
typedef __Float16x4_t float16x4_t;
        ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:43:9: error: unknown type name '__Float32x2_t'
typedef __Float32x2_t float32x2_t;
        ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:44:9: error: unknown type name '__Poly8x8_t'
typedef __Poly8x8_t poly8x8_t;
        ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:45:9: error: unknown type name '__Poly16x4_t'
typedef __Poly16x4_t poly16x4_t;
        ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:46:9: error: unknown type name '__Uint8x8_t'
typedef __Uint8x8_t uint8x8_t;
        ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:47:9: error: unknown type name '__Uint16x4_t'
typedef __Uint16x4_t uint16x4_t;
        ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:48:9: error: unknown type name '__Uint32x2_t'
typedef __Uint32x2_t uint32x2_t;
        ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:49:9: error: unknown type name '__Float64x1_t'
typedef __Float64x1_t float64x1_t;
        ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:50:9: error: unknown type name '__Uint64x1_t'
typedef __Uint64x1_t uint64x1_t;
        ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:51:9: error: unknown type name '__Int8x16_t'; did you mean '__int16_t'?
typedef __Int8x16_t int8x16_t;
        ^~~~~~~~~~~
        __int16_t
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/aarch64-linux-gnu/libc/usr/include/bits/types.h:38:26: note: '__int16_t' declared here
typedef signed short int __int16_t;
                         ^
/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.3.1/include/arm_neon.h:52:9: error: unknown type name '__Int16x8_t'
typedef __Int16x8_t int16x8_t;
        ^
[... cut ...]
ERROR: [SdsCompiler 83-5176] RTL template analysis exited with non-zero code processing /home/eva/Workspace2/CHai/src/software/swkernels/xi_eltwiseadd_top.cpp
sds++ log file saved as /home/eva/Workspace2/CHai/Release/_sds/reports/sds_xi_eltwiseadd_top.log
ERROR: [SdsCompiler 83-5004] Build failed

make: *** [src/software/swkernels/xi_eltwiseadd_top.o] Error 1
src/software/swkernels/subdir.mk:39: recipe for target 'src/software/swkernels/xi_eltwiseadd_top.o' failed

11:48:14 Build Finished (took 39s.668ms)

I have attached my Makefile. Where am i going wrong?
[Makefile.zip](https://github.com/Xilinx/CHaiDNN/files/3052904/Makefile.zip)

NP95, 2019-04-08:
For 2018.3, there is some issue with building from GUI. Just build from command line, it will work.

averr5, 2019-04-08:
@NP95 So how to proceed with command line build?

NP95, 2019-04-08:
Well just follow the command line build instructions which are right under
the instructions for building using GUI.

wlx20190529, 2020-03-17:
@NP95 I have the same problem as you. could you tell me how to solved it?

Okaymaddy, 2020-06-03:
@NP95 I am getting the following errors: 
[image]

I am using Sdx version 2019.1. How to fix this?

Okaymaddy, 2020-06-05:
@NP95 Do you also generate the custom platform? I mean is that necessary in order to build ChaiDNN?

meghuiyer, 2020-10-05:
15:40:48 **** Incremental Build of configuration Release for project test ****
make all 
make --no-print-directory pre-build
sdsoc_make_clean Release
 
make --no-print-directory main-build
Building target: test.elf
Invoking: SDS++ Linker
sds++ -L/usr/local/include/opencv4/opencv2 -L/tools/Xilinx/Vivado/2019.1/lnx64/tools/systemc/lib -L/tools/Xilinx/Vivado/2019.1/lnx64/tools/opencv/opencv_gcc -L/tools/Xilinx/SDx/2019.1/platforms/zed/sw/linux/boot -L/tools/Xilinx/SDK/2019.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi -L/tools/Xilinx/SDK/2019.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/arm-linux-gnueabihf/lib/opencv_build -L/tools/Xilinx/SDx/2019.1/xfopencv/include -sdcard ../src/data -Xlinker tools/Xilinx/SDK/2019.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi --remote_ip_cache /tools/Xilinx/Workspace/ip_cache -o "test.elf"  ./src/xf_otsuthreshold_accel.o ./src/xf_otsuthreshold_tb.o   -lopencv_core -lopencv_imgproc -lopencv_calib3d -lopencv_videoio -dmclkid 1 -mno-bitstream  -mno-boot-files  -sds-sys-config linux -sds-proc linux -sds-pf "zed" -emulation debug
Analyzing object files
... /tools/Xilinx/Workspace/test/Release/tools/Xilinx/SDK/2019.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi
... /tools/Xilinx/Workspace/test/Release/src/xf_otsuthreshold_accel.o
... /tools/Xilinx/Workspace/test/Release/src/xf_otsuthreshold_tb.o
Generating data motion network
INFO: [DMAnalysis 83-4494] Analyzing hardware accelerators...
INFO: [DMAnalysis 83-4497] Analyzing callers to hardware accelerators...
WARNING: [DMAnalysis 83-4492] Unable to determine the memory attributes passed to _src_mat.data of function w0_xf_OtsuThreshold at /tools/Xilinx/Workspace/test/src/xf_otsuthreshold_accel.cpp:38, please use mem_attribute pragma to specify
INFO: [DMAnalysis 83-4444] Scheduling data transfer graph for partition 0
INFO: [DMAnalysis 83-4446] Creating data motion network hardware for partition 0
INFO: [DMAnalysis 83-4448] Creating software stub functions for partition 0
INFO: [DMAnalysis 83-4450] Generating data motion network report for partition 0
INFO: [DMAnalysis 83-4454] Rewriting caller code
Skipping block diagram (BD), address map, port information and device registration for partition 0
Rewrite caller functions
Compile caller rewrite file /tools/Xilinx/Workspace/test/Release/_sds/swstubs/xf_otsuthreshold_accel.cpp
                 from /tools/Xilinx/Workspace/test/src/xf_otsuthreshold_config.h:40,
                 from /tools/Xilinx/Workspace/test/Release/_sds/swstubs/xf_otsuthreshold_accel.cpp:31:
                 from /tools/Xilinx/Workspace/test/Release/_sds/swstubs/xf_otsuthreshold_accel.cpp:31:
                 from /tools/Xilinx/Workspace/test/Release/_sds/swstubs/xf_otsuthreshold_accel.cpp:31:
                 from /tools/Xilinx/Workspace/test/src/xf_otsuthreshold_config.h:42,
                 from /tools/Xilinx/Workspace/test/Release/_sds/swstubs/xf_otsuthreshold_accel.cpp:31:
                 from /tools/Xilinx/Workspace/test/src/xf_otsuthreshold_config.h:42,
                 from /tools/Xilinx/Workspace/test/Release/_sds/swstubs/xf_otsuthreshold_accel.cpp:31:
                 from /tools/Xilinx/Workspace/test/Release/_sds/swstubs/xf_otsuthreshold_accel.cpp:31:
                 from /tools/Xilinx/Workspace/test/src/xf_otsuthreshold_config.h:42,
                 from /tools/Xilinx/Workspace/test/Release/_sds/swstubs/xf_otsuthreshold_accel.cpp:31:
/tools/Xilinx/SDx/2019.1/xfopencv/include/core/xf_math.h: In function ‘char xf::IdentifySignBits(ap_int<24>)’:
/tools/Xilinx/SDx/2019.1/xfopencv/include/core/xf_math.h:319:5: warning: label ‘signBitsLoop’ defined but not used [-Wunused-label]
     signBitsLoop:
     ^~~~~~~~~~~~
/tools/Xilinx/SDx/2019.1/xfopencv/include/core/xf_math.h: In function ‘short unsigned int xf::Inverse8(unsigned char, int, char*)’:
/tools/Xilinx/SDx/2019.1/xfopencv/include/core/xf_math.h:340:15: warning: unused variable ‘index’ [-Wunused-variable]
  unsigned int index;
               ^~~~~
/tools/Xilinx/SDx/2019.1/xfopencv/include/core/xf_math.h: In function ‘void xf::findInverse24(ap_int<24>&, int&, int&, unsigned int&, int*, int*)’:
/tools/Xilinx/SDx/2019.1/xfopencv/include/core/xf_math.h:380:12: warning: unused variable ‘N1’ [-Wunused-variable]
     int M1,N1,leftover_bits=0;
            ^~
/tools/Xilinx/SDx/2019.1/xfopencv/include/core/xf_math.h: In function ‘long unsigned int xf::DivideYByX24(ap_int24_t, ap_int24_t, int, int, int*, int*)’:
/tools/Xilinx/SDx/2019.1/xfopencv/include/core/xf_math.h:426:11: warning: unused variable ‘N2’ [-Wunused-variable]
  int N1=0,N2=0;
           ^~
                 from /tools/Xilinx/Workspace/test/Release/_sds/swstubs/xf_otsuthreshold_accel.cpp:31:
/tools/Xilinx/SDx/2019.1/xfopencv/include/imgproc/xf_otsuthreshold.hpp: In function ‘void xf::xfOtsuKernel(uint32_t (*)[256], uint16_t, uint16_t, uint8_t&)’:
/tools/Xilinx/SDx/2019.1/xfopencv/include/imgproc/xf_otsuthreshold.hpp:92:2: warning: label ‘HISTOGRAM_NORM_LOOP’ defined but not used [-Wunused-label]
  HISTOGRAM_NORM_LOOP:
  ^~~~~~~~~~~~~~~~~~~
/tools/Xilinx/SDx/2019.1/xfopencv/include/imgproc/xf_otsuthreshold.hpp:100:2: warning: label ‘SUM_LOOP’ defined but not used [-Wunused-label]
  SUM_LOOP:
  ^~~~~~~~
/tools/Xilinx/SDx/2019.1/xfopencv/include/imgproc/xf_otsuthreshold.hpp:107:2: warning: label ‘THRESHOLD_LOOP’ defined but not used [-Wunused-label]
  THRESHOLD_LOOP:
  ^~~~~~~~~~~~~~
                 from /tools/Xilinx/Vivado/2019.1/include/ap_int.h:54,
[... cut ...] Wl,--start-group    -lpthread -L/tools/Xilinx/SDx/2019.1/target/aarch32-linux/lib -lsds_lib -lxlnk_stub  -Wl,--end-group -o /tools/Xilinx/Workspace/test/Release/_sds/swstubs/test.elf '
sds++ log file saved as /tools/Xilinx/Workspace/test/Release/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

makefile:47: recipe for target 'test.elf' failed
make[1]: *** [test.elf] Error 1
makefile:39: recipe for target 'all' failed
make: *** [all] Error 2

Please help me to get through this error