Xilinx/CHaiDNN #143: CHaiDNN built on reVISION stack platform

hpollittsmith, 2019-01-22:
Hi,

Similar to some earlier posts (#22 and #30 ), I would like to build CHaiDNN on the reVISION platform (on ZCU102). I have followed the instructions from #30 to rename the HPx ports, and to not use HP0 and HP1 as these are used by the video interfaces in the PL. The warning is that performance may degrade for the CHaiDNN accelerator because it's only using 2 HP ports now. My question is, is it possible to use HP4 and HP5 in place of HP0 and HP1, or are they used by other resources as well?

Hugh

hpollittsmith, 2019-02-04:
I have built the CHaiDNN project on the reVISION Stack platform for the ZCU102, and I can run the GoogleNet application on the board. However, the output is not correct. The classId scores are all very small (0.003 or lower) for the sample images.

Has anyone successfully built and run CHaiDNN on reVISION Stack? Any suggestions or guidance would be appreciated.

guru3110, 2019-02-07:
Hi hpollitsmith,

I also had the same issue to built the chaidnn for reVision stack. I am using ZCU104 board. When ask the xilinx FAE team they told me that you can also be built using revision platforms with changes in port mappings.
It is not recommended because we can't build the higher DSP configs as the resources will be used by other IPs in revision platforms.

FAE team provide me the steps given below,
Please find the modifications to be done to be able to build CHaiDNN for zcu104 reVision platform below.
1) Find the system port names:
Use the command below to find the system port names for the platform
sds++ -sds-pf-info <path to platform>
2) Modify the port mappings
Open CHaiDNN/design/conv/include/xi_conv_config.h
Change the system port names in all the data_sys_port pragmas with the name obtained in step 1
Eg: #pragma SDS data sys_port(weights1:ps_e_S_AXI_HP0_FPD)   
3)  Map storage structures to URAMs
Set the below macros in xi_conv_config.h
#define XI_WTS_URAM_EN          1
#define XI_ISTG_URAM_EN          1
#define XI_OSTG_URAM_EN        1
4) Make sure that the platform has necessary clocks. 
        CHaiDNN employs DSP double pumping hence needs 2x clock. So if you are building the design for 100 MHz, make sure that the platform also has 200 MHz clock. Change the clock ids in the makefile accordingly.
5) Build the design
        Goto CHaiDNN/design/build and run make POOL_ENABLE=0 DECONV_ENABLE=0 to initiate the build.

Please try these steps it may work.

hpollittsmith, 2019-02-07:
Thanks, this is a nice summary. Note that on the zcu102 board there are no URAMs.

How did you modify the clock_ids in your design?

I have made some progress. If I compile the DIET_CHAI_ZUPLUS configuration, I can compile everything on the reVISION stack platform (making the changes to the sys_port pragmas) and run the application. The output looks correct.

However, if I then implement the Pool and Deconv blocks in hardware (POOL_ENABLE=1 and DECONV_ENABLE=1), although everything compiles and the application runs, the output is  incorrect (all very small values, even for the provided image samples). I'm going to try implement just Pool and/or just Deconv and see one of those configurations work.

guru3110, 2019-02-08:
Hi hpollittsmith,

Is it possible for you tell me the steps to compile for reVision stack platform. What changes you made in the make file. Please help me in the steps you modified for reVision stack it would be really helpful. 

I was able to build it for generic zcu104 platform and run the googlenet application. But the output of the inference is always same value for all images. Do you why this happening?

hpollittsmith, 2019-02-08:
I'll try to summarize the steps and modifications I made after I do a bit more testing. However, the cause of the problem I was having with the inference output basically came down to an oversight--make sure that any hardware configuration changes in xi_conv_config.h match up with modifications to the software\include\hw_settings.h.

This means the settings for XI_KER_PROC, XI_PIX_PROC, XI_ISTAGEBUFF_DEPTH, XI_OSTAGEFULL_DEPTH, and XI_WEIGHTBUFF_DEPTH.

guru3110, 2019-02-08:
Hi hpollitsmith,

Thanks for explaining me about the reason to the inference for the output.
I will also try from my side and update you regarding the status. Keep in
touch.