Xilinx/CHaiDNN #145: Coming CHaiDNN and reVISION stack software

hpollittsmith, 2019-02-12:
Hi,

Now that I have successfully built the CHaiDNN application (e.g., googlenet.elf) on the reVISION Stack platform, I'm wondering how to add the video pipeline software (such as gst_lib, video_lib and video_cmd) so that I can eventually have an application that grabs and frame, processes it through the CNN and then generates the output. Has anybody done this (or at least has a recommendation how this can be done)?

srinivasans74, 2019-02-28:
Hey i installed Chaidnn Hardware and i installed the software through make files. Where can i find the perfomance report of the compilation that i have done in the make file?.

Diezel66, 2019-06-29:
II also encountered this problem. Has anybody done this？