Xilinx/CHaiDNN #147: How to process one image at a time with the ChaiDNN Very?

mazhongzhong, 2019-02-20:


VishalX, 2019-02-20:
@mazhongzhong 
CHaiDNN-v2 design uses a batch of 2. 
You can run the design with one frame (Leave the other empty/un-initialized). But, In this way, you'll only utilize half of the compute capabilities of the design.

mazhongzhong, 2019-02-20:
@VishalX
We've tried leave the other empty/un-initialized, that'll end up with the wrong result of that single frame. So, for the time being, we have to duplicate one frame, then feed the two frames (the original one and the duplicated one) to the ChaiDNN at the same time, and throw away half of the results in the end. And the most frustrated fact is, throwing away half of the results is very time consuming. In our case, it takes about 170ms to complete the inference of our neural network model, and takes more than 500ms to unpack the two results (so the two results are continuous in the memory, respectively).

Thanks anyway.

And anyone know better ways to process videos with ChaiDNN v2?

cuongdv1, 2019-02-21:
I can process video as below flow
 
Open video file
LOOP
{
   grab frame 1
   grab frame 2
   normaliztion (frame1, frame2, outbuff)
   xiReadInput(outbuff, input)
   xiExe(input, output)
   unpack(output)  
}

mazhongzhong, 2019-02-21:
@cuongdv1
I have mentioned this flow. Unfortunately, it doesn't work for a lot of applications. For example, tracking objects in an video.

hpollittsmith, 2019-02-21:
Would you be able to share your code for reading in frames and running inference? I would like to do something similar.

srinivasans74, 2019-02-28:
Hey i installed Chaidnn Hardware and i installed the software through make files. Where can i find the perfomance report of the compilation that i have done in the make file?.