Xilinx/CHaiDNN #149: Error while building CHaiDNN software

averr5, 2019-02-21:
I have already built the hardware. Now I am trying to build the software following Build from source using the Makefile and I jump into the following errors : 
 
[log.txt](https://github.com/Xilinx/CHaiDNN/files/2888976/log.txt)

Any idea how to overcome these errors ?

VishalX, 2019-02-21:
It's because sds_lib.h not getting included. Make sure to set the correct SDx path in Makefile. 
 
An alternate way is to set the value of below macro to 0 in [scheduler.hpp](https://github.com/Xilinx/CHaiDNN/blob/master/software/scheduler/xi_scheduler.hpp).
#define LAYERWISE_PERFORMANCE

averr5, 2019-02-22:
I gave the right SDx path and I am trying to build the software again.
I am facing new errors now.
[error.txt](https://github.com/Xilinx/CHaiDNN/files/2894735/error.txt)

averr5, 2019-03-05:
@VishalX Any idea about the errors ?

VishalX, 2019-03-05:
`/../lib/gcc/aarch64-linux-gnu/7.3.1/../../../../aarch64-linux-gnu/bin/ld: cannot find -lxlnxdnn`

As you can see, it is not able to find the **libxlnxdnn.so** library. This .so is generated when you build the hardware. Review the path for this lib in your Makefile.

averr5, 2019-03-28:
I got the software build. I missed to copy libparser_arm.so, libxlnxdnn.so, libxstack.so files from SD_Card/zcu102/lib to SD_Card/lib

closed 2019-03-28 by averr5