Xilinx/CHaiDNN #150: Facing some errors when following instruction to build using SDx GUI with  ZCU104

ced1026, 2019-02-21:
I'm having trouble trying  ['Creating SDx GUI Progect'](https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md)

I got these errors.

I think it can not find **'__builtin_bit_part_select '** & **'__builtin_bit_part_set'**.
### Console Log
00:19:02 **** Incremental Build of configuration Release for project CHaiDNN_VGG ****
make pre-build main-build 
sdsoc_make_clean Release
' '
'Building file: ../src/software/scheduler/xi_perf_eval.cpp'
'Invoking: SDS++ Compiler'
sds++ -sds-hw XiConvolutionTop C:\Users\Dongwoo\workspace\CHaiDNN_VGG\src\design\conv\src\xi_convolution_top.cpp -clkid 1 -hls-tcl  C:\Users\Dongwoo\workspace\CHaiDNN_VGG\src\design\conv\scripts\config_core.tcl -sds-end -D__SDSOC -D__DECONV_ENABLE__ -D__POOL_ENABLE__ -D__SYNTHESIS__ -D__DSP48E2__ -D__CONV_ENABLE__ -Wall -O0 -I"../src" -I"C:\Users\Dongwoo\workspace\CHaiDNN_VGG\src\design\deconv\src" -I"C:\Users\Dongwoo\workspace\CHaiDNN_VGG\src\design\deconv\include" -I"C:\Users\Dongwoo\workspace\CHaiDNN_VGG\src\design\pool\src" -I"C:\Users\Dongwoo\workspace\CHaiDNN_VGG\src\design\pool\include" -I"C:\Users\Dongwoo\workspace\CHaiDNN_VGG\src\design\conv\src" -I"C:\Users\Dongwoo\workspace\CHaiDNN_VGG\src\design\conv\include" -I"C:\Users\Dongwoo\CHaiDNN-master\CHaiDNN-master\SD_Card\cblas\arm64\include" -I"C:\Users\Dongwoo\CHaiDNN-master\CHaiDNN-master\SD_Card\protobuf\arm64\include" -I"C:\Users\Dongwoo\CHaiDNN-master\CHaiDNN-master\SD_Card\opencv\arm64\include" -c -fmessage-length=0 -MT"src/software/scheduler/xi_perf_eval.o" -MMD -MP -MF"src/software/scheduler/xi_perf_eval.d" -MT"src/software/scheduler/xi_perf_eval.o" -o "src/software/scheduler/xi_perf_eval.o" "../src/software/scheduler/xi_perf_eval.cpp" -sds-hw XiDeconvTop xi_deconv_top.cpp  -files ../src/design/deconv/src/xi_deconv.hpp  -clkid 1 -sds-end -sds-hw PoolTop pooling_layer_dp_2xio_top.cpp  -files ../src/design/pool/src/pool_dp_2xio.hpp  -clkid 1 -sds-end -sds-sys-config a53_linux -sds-proc a53_linux -sds-pf "zcu104"
Analyzing source for RTL template usage
C:/Xilinx/Vivado/2018.3/include\ap_int_base.h:265:19: error: use of **undeclared identifier '__builtin_bit_select'**
    bool is_neg = _AP_ROOT_op_get_bit(reg.V, BITS - 1);
                  ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:622:39: note: expanded from macro '_AP_ROOT_op_get_bit'
#define _AP_ROOT_op_get_bit(Val, Bit) _ssdm_op_get_bit(Val, Bit)
                                      ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:529:23: note: expanded from macro '_ssdm_op_get_bit'
    bool __Result__ = __builtin_bit_select((void*)(&__Val2__), Bit); \
                      ^
C:/Xilinx/Vivado/2018.3/include\ap_int_base.h:268:13: error: use of **undeclared identifier '__builtin_bit_part_select'**
    exp.V = _AP_ROOT_op_get_range(reg.V, FLOAT_MAN, BITS - 2);
            ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:624:44: note: expanded from macro '_AP_ROOT_op_get_range'
#define _AP_ROOT_op_get_range(Val, Lo, Hi) _ssdm_op_get_range(Val, Lo, Hi)
                                           ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:547:5: note: expanded from macro '_ssdm_op_get_range'
    __builtin_bit_part_select((void*)(&__Result__), (void*)(&__Val2__), Lo, \
    ^
C:/Xilinx/Vivado/2018.3/include\ap_int_base.h:272:13: error: use of **undeclared identifier '__builtin_bit_part_select'**
    man.V = _AP_ROOT_op_get_range(reg.V, 0, FLOAT_MAN - 1);
            ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:624:44: note: expanded from macro '_AP_ROOT_op_get_range'
#define _AP_ROOT_op_get_range(Val, Lo, Hi) _ssdm_op_get_range(Val, Lo, Hi)
                                           ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:547:5: note: expanded from macro '_ssdm_op_get_range'
    __builtin_bit_part_select((void*)(&__Result__), (void*)(&__Val2__), Lo, \
    ^
C:/Xilinx/Vivado/2018.3/include\ap_int_base.h:277:13: error: use of undeclared **identifier '__builtin_bit_part_set'**
    man.V = _AP_ROOT_op_set_bit(man.V, FLOAT_MAN, 1);
            ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:623:45: note: expanded from macro '_AP_ROOT_op_set_bit'
#define _AP_ROOT_op_set_bit(Val, Bit, Repl) _ssdm_op_set_bit(Val, Bit, Repl)
                                            ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:538:5: note: expanded from macro '_ssdm_op_set_bit'
    __builtin_bit_part_set((void*)(&__Result__), (void*)(&__Val2__),   \
    ^
C:/Xilinx/Vivado/2018.3/include\ap_int_base.h:313:19: error: use of **undeclared identifier '__builtin_bit_select'**
    bool is_neg = _AP_ROOT_op_get_bit(reg.V, BITS - 1);
                  ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:622:39: note: expanded from macro '_AP_ROOT_op_get_bit'
#define _AP_ROOT_op_get_bit(Val, Bit) _ssdm_op_get_bit(Val, Bit)
                                      ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:529:23: note: expanded from macro '_ssdm_op_get_bit'
    bool __Result__ = __builtin_bit_select((void*)(&__Val2__), Bit); \
                      ^
C:/Xilinx/Vivado/2018.3/include\ap_int_base.h:316:13: error: use of **undeclared identifier '__builtin_bit_part_select'**
    exp.V = _AP_ROOT_op_get_range(reg.V, DOUBLE_MAN, BITS - 2);
            ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:624:44: note: expanded from macro '_AP_ROOT_op_get_range'
#define _AP_ROOT_op_get_range(Val, Lo, Hi) _ssdm_op_get_range(Val, Lo, Hi)
                                           ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:547:5: note: expanded from macro '_ssdm_op_get_range'
    __builtin_bit_part_select((void*)(&__Result__), (void*)(&__Val2__), Lo, \
    ^
C:/Xilinx/Vivado/2018.3/include\ap_int_base.h:320:13: error: use of **undeclared identifier '__builtin_bit_part_select'**
    man.V = _AP_ROOT_op_get_range(reg.V, 0, DOUBLE_MAN - 1);
[... 103 lines cut ...]
C:/Xilinx/Vivado/2018.3/include\ap_common.h:538:5: note: expanded from macro '_ssdm_op_set_bit'
    __builtin_bit_part_set((void*)(&__Result__), (void*)(&__Val2__),   \
    ^
C:/Xilinx/Vivado/2018.3/include\ap_int_base.h:685:18: error: use of undeclared identifier '__builtin_bit_select'
    return (bool)_AP_ROOT_op_get_bit(Base::V, i);
                 ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:622:39: note: expanded from macro '_AP_ROOT_op_get_bit'
#define _AP_ROOT_op_get_bit(Val, Bit) _ssdm_op_get_bit(Val, Bit)
                                      ^
C:/Xilinx/Vivado/2018.3/include\ap_common.h:529:23: note: expanded from macro '_ssdm_op_get_bit'
    bool __Result__ = __builtin_bit_select((void*)(&__Val2__), Bit); \
                      ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
Error while processing C:\Users\Dongwoo\workspace\CHaiDNN_VGG\src\software\scheduler\xi_perf_eval.cpp.
ERROR: [TemplateExtract 83-3509] Failed to run clang frontend action of rtl_template_extract!
ERROR: [SdsCompiler 83-5176] RTL template analysis exited with non-zero code processing C:/Users/Dongwoo/workspace/CHaiDNN_VGG/src/software/scheduler/xi_perf_eval.cpp
sds++ log file saved as C:/Users/Dongwoo/workspace/CHaiDNN_VGG/Release/_sds/reports/sds_xi_perf_eval.log
ERROR: [SdsCompiler 83-5004] Build failed

make: *** [src/software/scheduler/subdir.mk:29: src/software/scheduler/xi_perf_eval.o] Error 1

00:19:13 Build Finished (took 11s.427ms)

### Question
How can I resolve this? Is any other setting needed except for  instruction in ['Creating SDx GUI Progect'](https://github.com/Xilinx/CHaiDNN/blob/master/docs/BUILD_USING_SDX_GUI.md)?

nkpavan, 2019-02-22:
@ced1026 ,

As you are using 2018.3 tool, remove -D__SYNTHESIS__ symbol from the build settings.

ced1026, 2019-02-25:
Thanks for your help. so I resolved that problem!! but I got another error ut infra
### Problems
Description	Resource	Path	Location	Type
  [ skipping 2 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]	xi_convolution.hpp	/CHaiDNN_VGG/src/design/conv/src	line 3781	C/C++ Problem
  [ **skipping 2 instantiation** contexts, use -ftemplate-backtrace-limit=0 to disable ]	xi_convolution.hpp	/CHaiDNN_VGG/src/design/conv/src	line 6872	C/C++ Problem
  [ skipping 2 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]	xi_convolution.hpp	/CHaiDNN_VGG/src/design/conv/src	line 6885	C/C++ Problem
  [ skipping 3 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]	xi_convolution.hpp	/CHaiDNN_VGG/src/design/conv/src	line 2951	C/C++ Problem
  [ skipping 3 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]	xi_convolution.hpp	/CHaiDNN_VGG/src/design/conv/src	line 6755	C/C++ Problem
  [ skipping 7 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]	Release	/CHaiDNN_VGG	line 646, external location: C:\Xilinx\Vivado\2018.3\include\ap_int_ref.h	C/C++ Problem
make: *** [makefile:62: CHaiDNN_VGG.elf] Error 1	Release	/CHaiDNN_VGG	 	C/C++ Problem
SdsCompiler 83-5004: Build failed	Release	/CHaiDNN_VGG	 	C/C++ Problem
VPL 17-54: **The object 'run' does not have a property 'STEPS.OPT_DESIGN.ARGS.MORE  OPTIONS'.**	Release	/CHaiDNN_VGG	 	C/C++ Problem
VPL 60-341: Hardware accelerator integration failed. Aborting build_system. The following log file is available for debugging 'C:/Users/Dongwoo/workspace/CHaiDNN_VGG/Release/_sds/p0/vivado/vivado.log'. Contact your local Xilinx representative and provide the log file for further assistance.	Release	/CHaiDNN_VGG	 	C/C++ Problem
VPL 60-806: Failed to finish platform linker	Release	/CHaiDNN_VGG	 	C/C++ Problem

### Build Log
Attempting to get a license: ap_sdsoc
INFO: [VPL 17-86] Your ap_sdsoc license expires in 28 day(s)
Feature available: ap_sdsoc
INFO: [VPL 60-895]   Target platform: C:/Xilinx/SDx/2018.3/platforms/zcu104/zcu104.xpfm
INFO: [VPL 60-423]   Target device: zcu104
INFO: [VPL 60-1032] Extracting DSA to C:/Users/Dongwoo/workspace/CHaiDNN_VGG/Release/_sds/p0/vivado/.local/dsa
INFO: [VPL 60-251]   Hardware accelerator integration...
Creating Vivado project and starting FPGA synthesis.

WARNING: [VPL 60-1142] Unabled to read data from 'C:/Users/Dongwoo/workspace/CHaiDNN_VGG/Release/_sds/p0/vivado/output/generated_reports.log', generated reports will not be copied.

===>The following messages were generated while  creating FPGA bitstream. Log file: C:/Users/Dongwoo/workspace/CHaiDNN_VGG/Release/_sds/p0/vivado/vivado.log :
ERROR: [VPL 17-54] The object 'run' does not have a property 'STEPS.OPT_DESIGN.ARGS.MORE  OPTIONS'.

===>The following messages were generated while  creating FPGA bitstream. Log file: C:/Users/Dongwoo/workspace/CHaiDNN_VGG/Release/_sds/p0/vivado/vivado.log :
ERROR: [VPL 17-54] The object 'run' does not have a property 'STEPS.OPT_DESIGN.ARGS.MORE  OPTIONS'.
ERROR: [VPL 60-341] Hardware accelerator integration failed. Aborting build_system. The following log file is available for debugging 'C:/Users/Dongwoo/workspace/CHaiDNN_VGG/Release/_sds/p0/vivado/vivado.log'. Contact your local Xilinx representative and provide the log file for further assistance.
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling 'C:/Xilinx/SDx/2018.3/bin/vpl   --iprepo C:/Users/Dongwoo/workspace/CHaiDNN_VGG/Release/_sds/iprepo/repo  --iprepo C:/Xilinx/SDx/2018.3/data/ip/xilinx  --platform C:/Xilinx/SDx/2018.3/platforms/zcu104/zcu104.xpfm  --temp_dir C:/Users/Dongwoo/workspace/CHaiDNN_VGG/Release/_sds/p0  --output_dir C:/Users/Dongwoo/workspace/CHaiDNN_VGG/Release/_sds/p0/vpl  --input_file C:/Users/Dongwoo/workspace/CHaiDNN_VGG/Release/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels PoolTop:XiDeconvTop:XiConvolutionTop:adapter --webtalk_flag SDSoC  --remote_ip_cache C:/Users/Dongwoo/workspace/ip_cache --xp \"param:compiler.skipTimingCheckAndFrequencyScaling=1\" --xp "vivado_prop:run.impl_1.{STEPS.OPT_DESIGN.ARGS.MORE  OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp \"vivado_prop:run.impl_1.STEPS.PHYS_OPT_DESIGN.IS_ENABLED=1\" --xp "vivado_prop:run.impl_1.{STEPS.PHYS_OPT_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp "vivado_prop:run.impl_1.{STEPS.ROUTE_DESIGN.ARGS.MORE OPTIONS}={-directive Explore}" --xp \"vivado_prop:run.synth_1.{STEPS.SYNTH_DESIGN.TCL.PRE}={C:\Users\Dongwoo\workspace\CHaiDNN_VGG\src\design/src/conv/scripts/mcps.tcl}\" --xp \"vivado_prop:run.impl_1.{STEPS.PLACE_DESIGN.TCL.PRE}={C:\Users\Dongwoo\workspace\CHaiDNN_VGG\src\design/src/conv/scripts/mcps.tcl}\" --xp \"param:compiler.deleteDefaultReportConfigs=false\" '
sds++ log file saved as C:/Users/Dongwoo/workspace/CHaiDNN_VGG/Release/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

make: *** [makefile:62: CHaiDNN_VGG.elf] Error 1

17:24:20 Build Finished (took 1h:7m:52s.521ms)

### How can I resolve it?
Please help me one more... 
And the above log text is a part of full text.