Xilinx/CHaiDNN #166: Does CHaiDNN support the Depthwise convolution layer?

mazhongzhong, 2019-06-19:
The document (https://github.com/Xilinx/CHaiDNN/blob/master/docs/SUPPORTED_LAYERS.md) says the CHaiDNN supports the Depthwise Convolution Layer. But when we actually run a Depthwise Convolution Layer on it, the CHaiDNN can run it and outputs a result, but the result is not correct. So, I wonder Does the CHaiDNN support the Depthwise convolution layer or not? If the CHaiDNN does support the Depthwise convolution layer, has anyone used this successfully? Is there a demo to show how to use this feature?

ishida66, 2019-06-20:
It seems that the problem lies in the quantization rather than the layer implementation itself. 
https://arxiv.org/pdf/1803.08607.pdf

xlxxxxxl, 2019-07-02:
Hi @mazhongzhong @ishida66 
Have you solved this problem please?

Thanks
xuelei