Xilinx/CHaiDNN #181: Using Vitis to run ChaiDNN

nivi1501, 2022-07-21:
Hi, 
   Can you please share how to run ChaiDNN using Vitis tool? Vitis includes the functionality which was supported by SDSoc. Do I need to make significant changes in the code?

Regards,
Nivedita

nivi1501, 2022-07-22:
Ok, thanks a lot for the reply.
I just want to print statements in the console regarding which layer is getting executed. I do not care about synthesis and implementation results. Can you share fastest way to implement this using SDSoC development environment?

VishalX, 2022-07-28:
@nivi1501 There is one compile-time option that can be used to enable console logging. The scheduler already has all the messages.
Just build the software code with ENABLE_CONSOLE_TEXT macro enabled.

https://github.com/Xilinx/CHaiDNN/blob/master/software/scheduler/xi_scheduler.cpp#L521

Also, you can add custom messages as required.

nivi1501, 2022-07-28:
Thanks a lot.

closed 2022-07-28 by nivi1501